google-deepmind/mujoco #1379: [MJX] get_data: efc_J_rowadr shape mismatch

driesmarzougui, 2024-01-30:
Hi 

To render my MJX environment, I first use `mjx.get_data` to reallocate `mjx.Data` to an `mujoco.MjData`.
This however (sometimes) throws the following error:
[image]

My rendering currently works like [this](https://github.com/Co-Evolve/mujoco-utils/blob/new-framework/mujoco_utils/environment/base.py#L306), with [this](https://github.com/Co-Evolve/mujoco-utils/blob/new-framework/mujoco_utils/environment/mjx_env.py#L123) code implementing the model and data reallocation.

Any ideas on why this might be happening? If necessary, I can try to reproduce this issue using a minimal example. It however seems to only appear when I use non-random actions (i.e. actions from an optimised controller), making it a bit more difficult to provide a minimal example.

My XML model:  [BrittleStarMorphology.xml.zip](https://github.com/google-deepmind/mujoco/files/14096529/BrittleStarMorphology.xml.zip)

Thanks!

erikfrey, 2024-02-02:
Hi @driesmarzougui - first off, that's a cool star fish.

It's not clear to me why this error is happening from the report - I may need a smaller repro, thanks.  One clue though - in case it saves you some time - I think the only way this error could possibly happen is if you are passing in a different `mjx.Model` (with a different `m.nv`) than the one that was used to create the `mjx.Data` you are trying to get off device.  Do you think that's possible?

driesmarzougui, 2024-02-02:
Hi @erikfrey - thanks! I'm currently developing a bio-inspired robotics benchmark, which will surely include this brittle star model. 😉

Here are the requested values:
```
d_i.efc_J_rownnz.shape:  (52,)
d_i.efc_J_rowadr.shape:  (52,)
d_i.efc_J_colind.shape:  (2912,)
nefc:                    52
m.mv:                    56
```

I slightly reduced the number of DoFs of the model, so the error is currently:
[image]

erikfrey, 2024-02-02:
Oh yeah?  Explicitly casting to int makes the problem go away?  Great!  Want to open a PR?

driesmarzougui, 2024-02-05:
Done! https://github.com/google-deepmind/mujoco/pull/1401

closed 2024-02-06 by copybara-service[bot]