google-deepmind/mujoco #1383: Need Help to compute the kp and kv value theoretically for the Position and velocity servos ?

rishabkr29, 2024-02-01:
Hi,

I'm a student and I'm trying to use MuJoCo for my thesis. 

I'm looking for some help with the computation of kp and kv values for the position and velocity servo.

yuvaltassa, 2024-02-05:
What do you mean "theoretically"?

Do you have a real servo that you want to identify?

rishabkr29, 2024-02-05:
Sorry for not asking complete question, I am using a Position Servo at a slider joint and giving my reference value as position to it. so the system is failed to achieve to the desired position and we need to tuned the values of kp and kv, I wanted to understand the mathematical modelling behind the actuator section so that I can mathematically estimate the right values of kp and kv ?