google-deepmind/mujoco #1389: MJX performance on Unitree Go1

breakds, 2024-02-01:
I'm a research engineer working on robotics and I'm trying to use MJX to accelerate our training.

We used to run this [Unitree Go1 Model](https://github.com/google-deepmind/mujoco_menagerie/tree/main/unitree_go1) with a Threadripper 5975WX (32 core 64 threads), 5000 parallel environment. It finishes 160 x 5000 steps in less than 3 seconds, which translate to 267K steps per seconds.

I then tried it on 3.1.1 with MJX, with a dedicate GPU (Nvidia 4090) for simulation. After switching to `Newton`, setting `iterations=2` and `ls_itereations=2`, `eulerdamp=disable`, keep only collision between the feet and the floor, running it with 10K parallel environments and it finishes the same number of simulation steps in 400 seconds after initial compiling, which translates to only 4K steps per seconds. 

Is there anything else that I should do, or is this aligned with the expectation?

Thanks a lot!

erikfrey, 2024-02-02:
Hi @breakds,

Can you post the XML you used?  Here is what I just tried:

```
mjx-testspeed --mjcf=unitree/scene.xml --batch_size=8192 --unroll=4 --solver=newton
```

Here is the output I see:

```
Rolling out 1000 steps at dt = 0.002...

Summary for 8192 parallel rollouts

 Total JIT time: 26.59 s
 Total simulation time: 2.72 s
 Total steps per second: 3012235
 Total realtime factor: 6024.47 x
 Total time per step: 0.33 µs
```

This is on an A100 which is slower than a 4090 but has way more VRAM.

Here is a diff of what I changed in `go1.xml`:

```diff
4c4,6
<   <option cone="elliptic" impratio="100"/>
---
>   <option timestep="0.002" iterations="1" ls_iterations="4">
>     <flag eulerdamp="disable"/>
>   </option>
8,9c10,11
<       <geom friction="0.6" margin="0.001" condim="1"/>
<       <joint axis="0 1 0" damping="2" armature="0.01" frictionloss="0.2"/>
---
>       <geom friction="0.6" margin="0.001" condim="3"/>
>       <joint axis="0 1 0" damping="2" armature="0.01"/>
27c29
<         <geom group="3" type="capsule"/>
---
>         <geom group="3" type="capsule" contype="0" conaffinity="0"/>
62,63c64,65
<           <geom type="sphere" size="0.023" pos="0 0 -0.213" priority="1" solimp="0.015 1 0.031" condim="6"
<             friction="0.8 0.02 0.01"/>
---
>           <geom type="sphere" size="0.023" pos="0 0 -0.213" priority="1" solimp="0.015 1 0.031" condim="3"
>             friction="0.8 0.02 0.01" contype="0" conaffinity="1"/>
```

breakds, 2024-02-02:
What we have a modified version of that go1 xml. It looks like this:

```xml
<mujoco model="go1_description">
    <compiler angle="radian" meshdir="./meshes/" autolimits="true" />
    <size nconmax="100" />
    <option gravity='0 0 -9.806' iterations='2' ls_iterations="4"
            solver='Newton' timestep='0.005' impratio="100">
        <flag  eulerdamp="disable" />
    </option>
    <default>
        <geom contype="0" conaffinity="0" friction="0.6 0.3 0.3" rgba="0.5 0.6 0.7 1" margin="0.001" group="0"/>

        <motor ctrlrange="-33.5 33.5" ctrllimited="true"/>
        <camera fovy="60"/>
        <joint damping="1.0" armature="0.01" frictionloss="0.2" />

        <!-- The ctrlrange below are based on menagerie defaults and are just
             for reference. This will be overridden by the UnitreeGo1 class as
             a smaller range is more desired for the policy training purpose.
        -->
        <default class="actuator_on_hip">
            <position kp="50" ctrlrange="-0.802851 0.802851" forcerange="-23.7 23.7" />
        </default>
        <default class="actuator_on_thigh">
            <position kp="50" ctrlrange="-1.0472 4.18879" forcerange="-23.7 23.7" />
        </default>
        <default class="actuator_on_calf">
            <position kp="50" ctrlrange="-2.69653 -0.916298" forcerange="-35.55 35.55" />
        </default>
        <default class="foot">
            <!-- the 3rd value of solimp is roughly proportional to the deformation
                 of the foot when the robot stands on the ground. With this value (0.01)
                 the deformation is about 0.44 cm -->
            <geom type="sphere" size="0.02" pos="0 0 -0.213" priority="1" solimp="0.015 1 0.01"
                  condim="3" friction="0.8 0.02 0.01" rgba="0 0 0 1"
                  contype="1" conaffinity="1" />
        </default>
    </default>
    <asset>
        <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512"
                 height="3072"/>
        <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4"
                 rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
        <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5"
                  reflectance="0.2"/>
        <mesh name="trunk" file="trunk.stl" />
        <mesh name="hip" file="hip.stl" />
        <mesh name="thigh_mirror" file="thigh_mirror.stl" />
        <mesh name="calf" file="calf.stl" />
        <mesh name="thigh" file="thigh.stl" />
    </asset>

    <asset>
        <texture type="skybox" builtin="gradient" rgb1="1.0 1.0 1.0" rgb2="1.0 1.0 1.0" width="512" height="512"/>
        <texture name="plane" type="2d" builtin="flat" rgb1="1 1 1" rgb2="1 1 1" width="512" height="512" mark="cross" markrgb="0 0 0"/>
        <material name="plane" reflectance="0.0" texture="plane" texrepeat="3 3" texuniform="true"/>
    </asset>

    <worldbody>
        <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
        <geom name="floor" size="0 0 0.05" type="plane" material="groundplane" contype="1" conaffinity="1"/>

        <!-- <camera name="track" mode="trackcom" pos="0 -1.3 1.6" xyaxes="1 0 0 0 0.707 0.707"/> -->
        <body name="trunk" pos="0 0 0.325">
            <inertial pos="0.0116053 0.00442221 0.000106692" quat="0.0111438 0.707126 -0.00935374 0.706938" mass="4.801" diaginertia="0.0447997 0.0366257 0.0162187" />
            <freejoint/>
            <geom type="mesh" contype="0" conaffinity="0" group="1" rgba="0.913725 0.913725 0.847059 1" mesh="trunk" />
            <geom size="0.13 0.04675 0.057" type="box" rgba="0.913725 0.913725 0.847059 1" />
            <geom size="0.0005 0.0005 0.0005" pos="-0.01592 -0.06659 -0.00617" type="box" contype="0" conaffinity="0" group="1" rgba="0.8 0 0 0" />
[... 194 lines cut ...]
        <jointvel name="RR_calf_vel"    joint="RR_calf_joint" />
        <jointvel name="RL_hip_vel"     joint="RL_hip_joint"/>
        <jointvel name="RL_thigh_vel"   joint="RL_thigh_joint"/>
        <jointvel name="RL_calf_vel"    joint="RL_calf_joint"/>

        <accelerometer name="Body_Acc" site="imu"/>

        <gyro name="Body_Gyro" site="imu"/>

        <framepos name="Body_Pos" objtype="site" objname="imu"/>

        <framequat name="Body_Quat" objtype="site" objname="imu"/>

    </sensor>
    <keyframe>
        <key name="home"
             qpos="0 0 0.27 1 0 0 0 0 0.9 -1.8 0 0.9 -1.8 0 0.9 -1.8 0 0.9 -1.8"
             ctrl="0 0.9 -1.8 0 0.9 -1.8 0 0.9 -1.8 0 0.9 -1.8"/>
    </keyframe>
</mujoco>
```

Meanwhile I am trying to adapt your modification on it. Thanks @erikfrey !

Btw, I was looking for `mjx-testspeed` but cannot find it in my mjx installation. Is there a way I can do that as well? Thanks!

btaba, 2024-02-02:
Hi @breakds mjx-testspeed is available on the latest branch https://github.com/google-deepmind/mujoco/blob/main/mjx/mujoco/mjx/testspeed.py

breakds, 2024-02-02:
Thanks @btaba. I will compare it with my own benchmarking code and report back.

breakds, 2024-02-02:
With `testspeed`, the performance is now similar on Menagerie's model (tested on 3090)

```bash
$ python -m tools.mjx_speedtest --mjcf=go1.xml --batch_size=10000 --unroll=4 --solver=newton --nstep=400
```

```console
Summary for 10000 parallel rollouts

 Total JIT time: 21.76 s
 Total simulation time: 0.99 s
 Total steps per second: 4032338
 Total realtime factor: 8064.68 x
 Total time per step: 0.25 µs
 ```

I will need to dig into it to see what was the difference.

erikfrey, 2024-02-02:
Nice - those 4090 cards are quite good!  Closing, but feel free to reopen if you still think there's an issue in MJX.

closed 2024-02-02 by erikfrey