google-deepmind/mujoco #1393: Obtaining Gravity Bias

realSaddy, 2024-02-02:
Hi,

I am a student trying to calculate the individual biases on my model in simulation, specifically the gravity bias. mujoco_rne generates all of these combined, however, I want to do analysis on the gravity bias isolated.

I attempted to follow the suggestion from #1290, but I don't believe I'm doing it the correct way.
```
jacp = np.zeros((3, self.mujoco_model.nv), dtype=np.float64)
for i in range(self.mujoco_model.nbody):
    mujoco.mj_jacBody(self.mujoco_model, self.mujoco_data, jacp, None, i)
mass_vector = np.array([0, 0, -9.8 * mujoco.mj_getTotalmass(self.mujoco_model)])
mujoco_gravity = np.dot(mass_vector, jacp)
```

 Is there a recommended way of obtaining this with Mujoco?

Balint-H, 2024-02-02:
The end effector Jacobian lets you convert from joint space to task space. If you want to go from task space forces to joint space forces, you can use the transpose of that Jacobian. Also, you need to calculate the Jacobian of each body and the contribution of their weight separately.

``` python
import mujoco
import mujoco.viewer as viewer
import numpy as np

xml = './humanoid.xml'

def manual_grav_comp(model: mujoco.MjModel, data: mujoco.MjData):
    jacp = np.empty((3, model.nv))
    sum_qfrc = np.zeros((model.nv,))

    for i in range(1, model.nbody):
        mujoco.mj_jac(model, data, jacp=jacp, jacr=None, point=data.body(i).xipos, body=i)
        weight = model.opt.gravity * model.body(i).mass
        q_weight = jacp.T @ weight
        sum_qfrc -= q_weight
        # Note: The above could have been achieved using
        # mujoco.mj_applyFT(model, data, -weight, np.zeros(3), data.body(i).xipos, i, data.qfrc_applied)
        # It would have been faster due to using sparse jacobians internally.
    data.qfrc_applied = sum_qfrc

def load_callback(model=None, data=None):
    # Clear the control callback before loading a new model
    # or a Python exception is raised
    mujoco.set_mjcb_control(None)

    # `model` contains static information about the modeled system
    model = mujoco.MjModel.from_xml_path(filename=xml, assets=None)

    # `data` contains the current dynamic state of the system
    data = mujoco.MjData(model)

    if model is not None:

        # The provided "callback" function will be called once per physics time step.
        # (After forward kinematics, before forward dynamics and integration)
        # see https://mujoco.readthedocs.io/en/stable/programming.html#simulation-loop for more info
        mujoco.mj_forward(model, data)
        mujoco.set_mjcb_control(manual_grav_comp)

    return model, data

if __name__ == '__main__':
    viewer.launch(loader=load_callback)

```

yuvaltassa, 2024-02-02:
Thanks @Balint-H !

I think Balint's explanation and code show where the bugs are. Relatedly, the function [mj_jacSubtreeCom](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-jacsubtreecom) ([implementation](https://github.com/google-deepmind/mujoco/blob/7bb0ce4211f667e126ed7a844d8627f85b4985d9/src/engine/engine_support.c#L431)) does the adding and normalizing of the mass-weighted Jacobians for you. The subtree-com of the world body is the entire model, so if you had used that in your original code (no loop over bodies), then applying the force `total_mass * negative_gravity @ Jac` would have resulted in what you want.

closed 2024-02-02 by yuvaltassa

realSaddy, 2024-02-02:
Thank you @Balint-H & @yuvaltassa !

kevinzakka, 2024-10-02:
For future readers, I [benchmarked](https://colab.research.google.com/drive/1zlsplgSyk59hxnw3kOJMIxAXuwxXqOHD?usp=sharing) the three implementations and @yuvaltassa's suggestion is roughly 10x faster.

ppap36, 2024-10-19:
Sorry to bother you. I wonder whether there is a Code Auto-Completion on Mujoco(python).