google-deepmind/mujoco #1397: The movement of the bowl generated by the SDF plugin is not correct.

LinShuang22, 2024-02-03:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating my sphere model.

OS: Windows 11
MuJoCo Version: 3.1.1

I need to create a hollow sphere to serve as the shell for my simulation. I've noticed that the latest version of Mujoco supports and enables concave collision detection. And the official bowl example provided is similar to the model I am looking for. Therefore, I modified the parameters of the bowl to make it a closed shell.

It looks perfect, but there seems to be an issue with its movement. For instance, simply setting an x-direction velocity for the shell doesn't result in the expected movement. I compared it with a sphere created in the conventional manner. Additionally, when debugging in the viewer, the SDF model shakes every time I reload.

I'd like to know if this is an issue with the current version of SDF or if I cannot directly use the bowl as a shell.

Here is a screenshot / video, illustrating my question:
[image]

https://github.com/google-deepmind/mujoco/assets/119031680/e2030f79-abd4-45fa-b3c9-2c158024cc0e

Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="shell">
  <extension>
    <plugin plugin="mujoco.sdf.bowl">
      <instance name="bowl">
        <config key="height" value="0.041"/>
        <config key="radius" value="0.041"/>
        <config key="thickness" value="0.001"/>
      </instance>
    /
    </plugin>
  </extension>
  
  <visual>
    <scale camera="0.1" contactwidth="0.1"
           framelength="0.2" framewidth="0.02"
           jointlength="0.2" jointwidth="0.02"
    />
    <rgba camera="1 0 0 0.6" />
  </visual>

  <asset>
    <mesh name="bowl">
      <plugin instance="bowl"/>
    </mesh>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <option sdf_iterations="5" sdf_initpoints="20" gravity="0 0 -9.8" timestep="0.002" />

  <default>
    <geom solref="0.01 1" solimp=".95 .99 .0001" condim="6" />
  </default>

  <statistic meansize=".1"/>

  <worldbody>
    <light name="left" pos="0 0 4"/>
    <light name="right" pos="-1 -1 2" dir="1 1 -2"/>
    <geom name="ground" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>

<!--     SDF bowl, a shell -->
    <body name="shell" euler="0 0 0" pos="0 0 0.041">
      <joint name="shell_joint" pos="0 0 0" type="free"/>
      <inertial pos="0 0 0" mass="5.0e-02" diaginertia="5.0e-05 5.0e-05 5.0e-05"/>
      <geom type="sdf" name="shell_geom" mesh="bowl" rgba=".2 .8 .8 0.5">
        <plugin instance="bowl"/>
      </geom>
    </body>

    <body name="test_ball" pos="0 0.1 0.041">
      <joint type="free" pos="0 0 0" />
      <inertial pos="0 0 0" mass="5.0e-02" diaginertia="5.0e-05 5.0e-05 5.0e-05"/>
      <geom name="ball_geom" type="sphere" size="0.041"/>
    </body>
  </worldbody>

  <contact>
    <pair name="shell1-ground" geom1="ground" geom2="shell_geom" friction="1 1 0.9 0.2 0.2" />
  </contact>

</mujoco>
```

</details>

yuvaltassa, 2024-02-05:
Is your freejoint at the CoM? If it isn't then you'll get something that looks like this.

saran-t, 2024-02-06:
If I recall correctly the bowl SDF only generates contact points on the inner (concave) surface, so it's not suitable for use as a model for a hollow ball.

LinShuang22, 2024-02-06:
> If I recall correctly the bowl SDF only generates contact points on the inner (concave) surface, so it's not suitable for use as a model for a hollow ball.

Really? But I can see contact points between the outer surface of the bowl and the ground.
[image]

LinShuang22, 2024-02-06:
> If I recall correctly the bowl SDF only generates contact points on the inner (concave) surface, so it's not suitable for use as a model for a hollow ball.

Really? But I can see contact points between the outer surface of the bowl and the ground.
[image]

If the bowl is not suitable as a model for a hollow sphere, should I write a plugin for a hollow sphere myself?

quagla, 2024-02-15:
What did you set the attributes to? Could it be that you set the `height` very close to the surface?

LinShuang22, 2024-02-16:
> What did you set the attributes to? Could it be that you set the `height` very close to the surface?
> 
> EDIT: I just saw the XML, I would try to set it to a large number so that there is no gradient discontinuity near the surface. Let me know if it helps!

@quagla Sorry, I didn't understand your meaning. What does "height" refer to? What I need is continuous contact both inside and outside the spherical shell, achieving collisions in both cases.

This is the shape of my robot, and it achieves the motion of the spherical shell through the driving of the internal car.
[image]

quagla, 2024-02-20:
@LinShuang22 The SDF plugin you are using is for a cut hollow sphere. The "height" is the position of the cut. You have set it to the radius to effectively the plane that cuts the sphere is tangent to it and you get a full sphere. 

However I'm afraid that the plane passes exactly thought the contact point and it could cause a discontinuity in the gradient of the SDF, which may cause some issues with finding the contact points. 

This is just a theory but it's easy to check if it is true by setting height to a high number, so the cutting plane is far away and has no influence on the collision.

LinShuang22, 2024-02-22:
> @LinShuang22 The SDF plugin you are using is for a cut hollow sphere. The "height" is the position of the cut. You have set it to the radius so effectively the plane that cuts the sphere is tangent to the surface and you get a full sphere.
> 
> However I'm afraid that the plane also passes exactly thought the contact point and it could cause a discontinuity in the gradient of the SDF, which may cause some issues with finding the contact points.
> 
> This is just a theory but it's easy to check if it is true by setting height to a high number, so the cutting plane is far away and has no influence on the collision.

@quagla  Thank you. I tried increasing the `height`, but it doesn't seem to have any effect. Additionally, I tried increasing the size of `sdf_initpoints`, and this seems to have a significant effect. 

When debugging the model, a larger `sdf_initpoints` seems to lead to more stable simulation. However, when I attempt to control the shell, it doesn't seem to have an effect, and it deviates from the desired trajectory, similar to the video on this page.

I would like to know if there is an issue with the current version of the SDF functionality or if there is a problem with the plugin I am using.  Should I write a hollow sphere plugin for my model?
[image]

RafaelsNeurons, 2024-02-27:
You can also make the sdf custom for your custom mesh, look how they have done it in the cow example https://github.com/google-deepmind/mujoco/blob/main/model/plugin/sdf/cow.xml

Also you can visualize the SDF of each object: Rendering> Toggle "SDF iters"

Hope that helps you :)

LinShuang22, 2024-03-05:
@RafaelsNeurons Thank you :) . I tried the methods you provided, but it still seems unstable. I'm using an STL file for the mesh, and Mujoco allows up to 200,000 faces. I exported the STL file as finely as possible, but the collision between the spherical shell and the ground remains unstable, perhaps due to too few contact points between the shell and the ground.
[image]

saran-t, 2024-03-05:
I'm still reasonably convinced that you can simulate this robot without modeling its internal mechanism in the same way as it's actually implemented in real hardware (and therefore not needing the hollow sphere at all).

However, if you really want to build out the hollow sphere you could try the following construction.
- Put a normal sphere and the bowl SDF in the same body with the same size and position.
- Set the contype/conaffinity such that the bowl only generates contacts with the internal mechanism and vice versa, while the primitive (convex) sphere takes care of contacts with the rest of the world.

This should work around unstable SDF contacts with the ground while still allowing you to use the hollow sphere to model the movement system.

LinShuang22, 2024-03-05:
> I'm still reasonably convinced that you can simulate this robot without modeling its internal mechanism in the same way as it's actually implemented in real hardware (and therefore not needing the [...]

Thank you for your suggestions. I will try that.

closed 2024-03-17 by LinShuang22