google-deepmind/mujoco #1398: System Identification

QiyangYan, 2024-02-03:
**How to ask for help**

Hi,

I'm a student and I'm trying to use MuJoCo for Sim2Real of a gripper.

I'm looking for some help with System Identification.

You mentioned the guidelines for system identification will be released later in 2023, I'm wondering where I can find it. Thanks!

yuvaltassa, 2024-02-05:
Unfortunately sysID best-practices are not yet published, but @kevinzakka and I are finally making some progress. Thank you for motivating us to finish this 🙂.

TL;DR you want to make a box-constrained least squares problem that minimizes (real_data - simulated_data) as a function of the model parameters, subject to physicality constraints (e.g. no negative damping etc).

You probably want to adjust damping, frictionloss, armature and gain.

Closing for now, hopefully will have updates soon.

closed 2024-02-05 by yuvaltassa

ykim104, 2024-10-25:
Do you have updates on this? Thanks in advance for your reply! :) @yuvaltassa

yuvaltassa, 2024-10-27:
Unfortunately no 😞 
It's still on our to-do list!
This will happen.

Stocko-2073, 2026-01-28:
i'm hoping this is the year 🤞