google-deepmind/mujoco #1399: Modeling badminton racket as an elastic rod using cable

alex1yaremchuk, 2024-02-04:
### Discussed in https://github.com/google-deepmind/mujoco/discussions/1386

Hi,

I'm an amateur researcher and I'm trying to use MuJoCo for modeling a badminton racket (later with an arm).
I'm not a physicist but I'm a programmer.

I'm looking for some help with cable composite's parameters for elasticity.

I'm aiming at simulating a flexible rod (immitating badminton racket from the real world). 

Questions are: 
1) Is the model correct in general in terms of stability (I got an advice of adding stiffness to main joint which helped a lot)?
Maybe something can be made simpler. 
My aim is not in rendering but in measuring model's params (stick end's velocity).

2) How to regulate rod's elasticity using these related parameters? 

- composite.bend
- main joint.stiffness

My current model is available at this colab at 
https://colab.research.google.com/drive/1TinjgjJFXFqX_hrFsZz1fFQpYQVsjCKH#scrollTo=0jBvlH6yAhA9&line=45&uniqifier=1

quagla, 2024-03-04:
Hi @alex1yaremchuk sorry for the delay! 

The cable model is in general very stable because it's based on reduced coordinates. I would suggest to ignore the main joint stiffness, why do you need it at all? Just use the plugin parameters (bend and twist) see here for more details https://github.com/google-deepmind/mujoco/blob/main/plugin/elasticity/README.md

alex1yaremchuk, 2024-03-05:
Thank you, guys! I'll give it a try. 

the delay is ok with me - this project is totally facultative - it can wait.

quagla, 2024-03-06:
Glad it helped!

closed 2024-03-06 by quagla