google-deepmind/mujoco #1400: [`MJX`] how to map `body.name` to `body.tid`

Kallinteris-Andreas, 2024-02-05:
Hi,

I'm a maintainer of [Gymnasium](https://gymnasium.farama.org/) & the project manager of [Gymnasium-Robotics](https://robotics.farama.org/), and I'm trying to use `MuJoCo-MJX` for "prototyping `MJX`-based RL environments in `Gymnasium`, `Gymnasium-Robotics`, `Metaworld`, `MO-Gymnasium`".

I want the ID of a body part while knowing its name

with `mujoco` you can get the ID of a body part with:
```py
data.body("torso").id
```

how can I do the same thing with MJX

Thanks!

btaba, 2024-02-05:
Hi @Kallinteris-Andreas , for now you can use MuJoCo to get the id, and then use the id with MJX. Check out the [colab](https://colab.research.google.com/github/google-deepmind/mujoco/blob/main/mjx/tutorial.ipynb) where the BarkourEnv has examples of this

closed 2024-02-05 by Kallinteris-Andreas

varadVaidya, 2024-02-11:
hi @Kallinteris-Andreas , I am also in the process of converting my current custom MuJoCo Gymnasium environment to use MJX.
If possible can you share or point to a Minimal working example for such workflow. I am currently trying to take inspiration from MjxEnv in brax (as in the colab tutorial), but would love to see some other examples as well

Kallinteris-Andreas, 2024-02-11:
@varadVaidya Keep it is WIP, but you could check out https://github.com/Farama-Foundation/Gymnasium/pull/834