google-deepmind/mujoco #1403: How to compile examples using a CMakeLists for Mujoco

nimesh00, 2024-02-05:
Hi, I'm trying to make a CMakeLists.txt to compile my simulation file. It is very similar to basic.cc but is in other directory. I have inflated the mujoco tar ball and set and environment variables for mujoco include files and libraries. These are my corresponding Makefile and CmakeLists.txt that I'm using to compile.

Makefile
`COMMON=-O2 -I../include -L${MUJOCO_LIBRARY_DIR} -std=c++17 -pthread -Wl,-no-as-needed -Wl,-rpath,${MUJOCO_LIBRARY_DIR}

.PHONY: all
all:
	$(CXX) $(COMMON) ../src/mj_sim.cpp          -lmujoco -lglfw         -o mj_sim`

CMakeLists.txt
`cmake_minimum_required(VERSION 3.5)
project(mujoco_sim_test)
set(USE_GL 1)

set(CMAKE_CXX_STANDARD 17)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -O2 -pthread -Wl,-no-as-needed -Wl,-rpath,${MUJOCO_LIBRARY_DIR}")

include_directories(
    include
)
link_directories(${MUJOCO_LIBRARY_DIR})

add_executable(mj_sim
    src/mj_sim.cpp
)
target_link_directories(${MUJOCO_LIBRARY_DIR})
# target_link_libraries(${BIN_NAME} ${MUJOCO_LIBRARY_DIR}/libmujoco.so)
target_link_libraries(mj_sim mujoco glfw)`

I am using them separately (in different build directories just to test compilation).

`make` with Makefile is working fine. But using `make` with CMakeLists.txt is not able to find the mujoco library (cannot find -lmujoco). Can anyone point out what I'm doing wrong or tell me the proper way of using Mujoco library in C++ projects?
Thanks.

ZLWSSS, 2024-03-28:
Hi nimesh00, you could set the relative path of mujoco library in Link_directories. The reason why you can not find it probably is that you have not installed mujoco.

traversaro, 2024-03-28:
If `MUJOCO_LIBRARY_DIR` is an env variable, to use it in CMake you need to write `$ENV{MUJOCO_LIBRARY_DIR}`, not `${MUJOCO_LIBRARY_DIR}`. A possibly easier alternative is that you compile mujoco from source using CMake, in that case the install directory will contain a `mujoco-config.cmake` file  (not present in the released binary archives, as far as I know) that will permit to link mujoco simply with:

~~~cmake
find_package(mujoco REQUIRED)

target_link_libraries(<yourtarget> mujoco::mujoco)
~~~

nimesh00, 2024-03-28:
Thanks, @ZLWSSS @traversaro for the reply. I managed to figure it out eventually. I compiled it from source and specified the library path to CMake by appending the installation path to `CMAKE_PREFIX_PATH` in my CMakeLists.txt. Closing this issue now.

closed 2024-03-28 by nimesh00