google-deepmind/mujoco #1404: Modelling a ratcheting mechanism

Jerold-S, 2024-02-06:
**How to ask for help**

Hi,

I am trying to model a ratcheting mechanism between two rotating bodies and using the Mujoco python bindings. My idea to do this was to use an equality contraint and toggle it on and off using the measurement of the joint speeds and torque application as the triggers. I thought this would be possible using the `active0` attribute of the eqaulity constraint, however this does not seem to work.

Below is a similified python file that includes the xml definition with two adjacent cylinders. One recieves a sinusoidal velocity target, while the other is linked to it through a joint equality contraint. 

<details>
  <summary>Python Script</summary>

```Python
import mujoco
import numpy as np
import mediapy as media
import matplotlib.pyplot as plt

xml = """
<mujoco>
    <asset>
        <texture name="grid_blu" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="500" height="500" mark="edge" markrgb=".4 .5 .6"/>
        
        <texture name="grid_red" type="2d" builtin="checker" rgb1=".3 .2 .1" rgb2=".4 .3 .2" width="500" height="500" mark="edge" markrgb=".4 .5 .6"/>

        <material name="grid1" texture="grid_blu" texrepeat="1 1" texuniform="true" reflectance=".2"/>

        <material name="grid2" texture="grid_red" texrepeat="2 2" texuniform="true" reflectance=".2"/>
    </asset>

    <worldbody>
        <body name="cyl_1" pos="0 .12 0" >
            <joint name="axis1" type="hinge" axis="0 1 0" damping = ".2" />
            <geom type="cylinder" size=".5 .1" euler="90 0 0" material="grid1" />
        </body>

        <body name="cyl_2" pos="0 -.12 0" >
            <joint name="axis2" type="hinge" axis="0 1 0" damping = ".2" />
            <geom type="cylinder" size=".5 .1" euler="90 0 0" material="grid2" />
        </body>
    </worldbody>

    <equality>
        <joint name="ratchet" active="true" joint1="axis1" joint2="axis2" />
    </equality>

    <actuator>
        <velocity name="inLoad" joint="axis1" kv="100" />
    </actuator>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)

duration = 3
dt = model.opt.timestep = 0.001
nsteps = int(duration/dt)
time = np.linspace(0, duration, nsteps)

frames=[]

speeds = np.zeros([nsteps, 2])

for i, t in enumerate(time):
    mujoco.mj_step(model, data)
    
    data.ctrl = 10 * np.sin(np.pi * 2 * data.time)

    speeds[i] = np.array([data.joint("axis1").qvel[0], data.joint("axis2").qvel[0]])
    
    if data.joint("axis1").qacc[0] < 0: 
        model.equality("ratchet").active0[0] = 0
    elif data.joint("axis1").qvel[0] >= data.joint("axis2").qvel[0]:
        model.equality("ratchet").active0[0] = 1
        
plt.plot(time, speeds)
```

</details>

I have tried this with the equality constraint `active` attribute set to true and false in the initial model definition, but either way, the toggling of the contraint does not seem to work, the true case it never deactivates, and in the false case, it never activates. Is there another way to do this, or am I missing something in the equality constraint?

Cheers

Jerold-S, 2024-03-05:
After some time working through the MuJoCo documentation and some trial and error I think i have worked this out. As there has been no response, I presume that no-one is currently interested in ratcheting/clutch mechanisms, but I will try to document my progress in case of future interest.

Regarding the toggling of the equality constraint - I was trying to toggle using the `active0` in the `data.equality` structure, however the documentation fairly clearly states that equalities should be toggled using `data.eq_active`. I clearly glossed over this...

After correcting that, it was clear that the joint equality constraint looks at the difference in positions between two joints, not the difference in velocities. This caused issues when toggling the constraint on after some differential movement between the two connected bodies, as they would try to spring back to where they were at the start of the simulation. To avoid this, the first value of the `polycoef` attribute of the equality needed to be updated each time the equality was toggled on. I worked out this value was stored in `model.equality("equality_name").data[0]` field.

The one question I have is whether this is the correct way to update the equality referece position? I know the advice is not to modify anything within the `model` structure at runtime, though this seems to work. Maybe @yuvaltassa can answer or can pass on to someone who can?

<details>
<summary>
Here is an  updated example where there is a ratchet/1-way clutch between two bodies - 1 driven by an actuator and the other on a damped axis.
</summary>

```Python
import numpy as np
import mediapy as media
import matplotlib.pyplot as plt
import mujoco

from math import copysign

damping = 20

xml = f"""
<mujoco>
    <asset>
        <texture name="grid_blu" type="2d" builtin="checker" rgb1=".1 .2 .3" rgb2=".2 .3 .4" width="500" height="500" mark="edge" markrgb=".4 .5 .6"/>
        
        <texture name="grid_red" type="2d" builtin="checker" rgb1=".3 .2 .1" rgb2=".4 .3 .2" width="500" height="500" mark="edge" markrgb=".4 .5 .6"/>

        <material name="grid1" texture="grid_blu" texrepeat="1 1" texuniform="true" reflectance=".2"/>

        <material name="grid2" texture="grid_red" texrepeat="2 2" texuniform="true" reflectance=".2"/>
    </asset>

    <worldbody>
        <body name="cyl_1" pos="0 .12 0" >
            <joint name="axis1" type="hinge" axis="0 1 0" />
            <geom type="cylinder" size=".5 .1" euler="90 0 0" material="grid1" />
        </body>

        <body name="cyl_2" pos="0 -.12 0" >
            <joint name="axis2" type="hinge" axis="0 1 0" damping = "{damping}" />
            <geom type="cylinder" size=".5 .1" euler="90 0 0" material="grid2" />
        </body>
    </worldbody>

    <equality>
        <joint name="ratchet" active="false" joint1="axis1" joint2="axis2"
        polycoef="0 1 0 0 0" />
    </equality>

    <actuator>
        <velocity name="inLoad" joint="axis1" kv="20000" />
    </actuator>
</mujoco>
"""

model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)
renderer = mujoco.Renderer(model)

framerate = 30
duration = 6
dt = model.opt.timestep = 0.001
nsteps = int(duration/dt)
time = np.linspace(0, duration, nsteps)

frames=[]

speeds = np.zeros([nsteps, 2])
accels = np.zeros([nsteps, 2])
ratchet = np.zeros([nsteps]) - 1

latch = 0

for i, t in enumerate(time):
    mujoco.mj_step(model, data)
    
    data.ctrl = 3* np.sin(np.pi * data.time)

    speeds[i] = np.array([data.joint("axis1").qvel[0], data.joint("axis2").qvel[0]])
    accels[i] = np.array([data.joint("axis1").qacc[0], data.joint("axis2").qacc[0]])
    
    if ( # Disengage
        data.joint("axis1").qacc[0] <= 
        data.joint("axis2").qvel[0] * damping / model.body("cyl_2").inertia[2] *  -copysign(1, data.joint("axis2").qvel[0])
        and latch == 1
        ):
        data.eq_active = ratchet[i] = 0
        latch = 0        
    
    elif ( # Engage
        data.joint("axis1").qvel[0] >= data.joint("axis2").qvel[0]
        and latch ==0
        ):
        data.eq_active = ratchet[i] =  1
        model.equality("ratchet").data[0] = (
            data.joint("axis1").qpos[0] - data.joint("axis2").qpos[0])
        latch = 1

    if len(frames) < data.time * framerate:
        renderer.update_scene(data)
        frames.append(renderer.render())

# Plotting Behaviour
fig, ax = plt.subplots(2, 1, sharex=True)

ax[0].plot(time, speeds)
ax[0].set_ylabel("Angular \nVelocity [rad/s]")
ax[0].set_xlabel("Time [s]")

ax0 = ax[0].twinx()
ax0.step(time, ratchet, "red", linewidth="1")
ax0.set_yticks([-1, 0, 1], ["No Action", "OFF", "ON"])
ax0.set_ylabel("Ratchet Toggle")

ax[1].plot(time, accels)
ax[1].set_ylabel("Angular \nAcceleration [rad/s/s]")
ax[1].set_xlabel("Time [s]")
ax[1].legend(["Drive cylinder", "Driven Cylinder"])

media.show_video(frames, fps=framerate)
```
</details>

yuvaltassa, 2024-03-06:
Sorry, totally missed the OP.

- Yes, `mjData.eq_active` is what you want (note the `0` suffix has meaning (like `mjModel.qpos0`) and [inline](https://github.com/google-deepmind/mujoco/blob/e2c14188adc8f6170aaf03a2f9c3424ab191d629/include/mujoco/mjmodel.h#L953) [documentation](https://github.com/google-deepmind/mujoco/blob/e2c14188adc8f6170aaf03a2f9c3424ab191d629/include/mujoco/mjdata.h#L226) is useful, not to mention the [actual documentation](https://mujoco.readthedocs.io/en/latest/computation/index.html#user-inputs))
- Yes, MuJoCo does not currently support constraints on velocities, but there is no blocker in principle. However it would be a not-small project to add this feature.
- Given the previous point, your current solution is the correct one :slightly_smiling_face: 

Finally, the advice is not exactly "not to modify anything within the model structure at runtime" but rather "not to modify anything within the model structure at runtime _unless you know what you are doing_" :stuck_out_tongue:  Modifying `mjModel.eq_data` is a legitimate thing to do as it has no side effects, the main caveat is that the semantics of this field are rather tricky to understand, so we prefer people with some general understanding to be doing this. Clearly you are such a person.

closed 2024-03-06 by yuvaltassa