google-deepmind/mujoco #1405: `mujoco==3.1.2` gives different `model.nbvh` than different versions

Kallinteris-Andreas, 2024-02-07:
`mujoco==3.1.2` gives a different number of bounding volumes in all bodies

```sh
$ pip list | grep mujoco
mujoco                        3.1.1
mujoco-mjx                    3.1.1
$ py                    
>>> import mujoco
>>> model = mujoco.MjModel.from_xml_path("pusher.xml")
>>> model.nbvh
18
```

```sh
$ pip list | grep mujoco
mujoco                        3.1.2
mujoco-mjx                    3.1.2
$ py                    
>>> import mujoco
>>> model = mujoco.MjModel.from_xml_path("pusher.xml")
>>> model.nbvh
8
```

asset xml_file: https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/assets/pusher.xml

- Include the following context:
  - Operating system.
  - MuJoCo version (and if the bug is new, the version where it used to work).
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)?

yuvaltassa, 2024-02-07:
Why do you consider this a bug?

Kallinteris-Andreas, 2024-02-07:
Hey,
The changelog for 3.1.2 did not indicate that anything would change regarding default bounding volumes, if you believe this to be intended behavior feel free to close the issue

yuvaltassa, 2024-02-07:
I'll let @quagla comment on the cause of the change (maybe it is a problem!) but more generally the changelog cannot report everything that happened, for that you have the commit history. We tend to report things that the user could care about, and the BV trees are a very internal thing that users generally don't know or care about.

That said, bringing a maybe-bug to our attention is always welcome 🙂

So thanks for that!

@quagla, please check if this is benign or not?

saran-t, 2024-02-13:
This isn't being picked up by our internal golden tests (admittedly we only have two test cases that print out `nbvh`). Definitely worth checking just to be sure.

quagla, 2024-02-15:
Do you get a different behavior also if you do NOT set contype=0 and conaffinity=0? I have refactored the BVH allocation between 3.1.1 and 3.1.2 when working on `discardvisual` because it was needed for that change. I see that all geoms are visual in the swimmer so I think the new behavior (nbvh=0) is the correct one.

Kallinteris-Andreas, 2024-02-15:
Without `contype=0` and `conaffinity=0` swimmer gets
```
>>> model.nbvh
4
```
regardless of `mujoco` version (`3.1.1`, `3.1.2`)

quagla, 2024-02-15:
Great! So I mark it as solved. The intended behavior is that the BVH should ignore visual geoms and it used to be a bug that it didn't. Thanks for flagging this!