google-deepmind/mujoco #1407: support for visual/⁠global/lookat

vikashplus, 2024-02-07:
**Is your feature request related to a problem? Please describe.**
Better specification of default camera to control the opening view of a model. Currently, we only support `visual/⁠global/azimuth` and `visual/⁠global/elevation`

**Describe the solution you'd like**
support `visual/⁠global/lookat`

**Additional context**
Say you have a large house scene and you want to control which room should we start in. Or maybe it should be the entrace. This feature will enable this capability.

yuvaltassa, 2024-02-09:
This is vale controlled by `<statistic center/>`

yuvaltassa, 2024-02-09:
[Documentation](https://mujoco.readthedocs.io/en/latest/XMLreference.html#statistic-center)

closed 2024-02-09 by yuvaltassa

vikashplus, 2024-02-09:
Exactly what I needed. Thanks @yuvaltassa 

Should we add a line in the doc next to `visual/⁠global/elevation/azimuth`
> The lookat point can be specified using `statistic/ center`

commit 4ad1dc4b (2024-02-09): Mention the roles of "center" and "extent" in visual/global/{azimuth,elevation}. Fixes #1407.

PiperOrigin-RevId: 605649532
Change-Id: I0ca0142a3c164a502fefb6b91d204c4bb3d25a95