google-deepmind/mujoco #1410: user_scn and update_hfield() equivalents in Renderer class

lok-i, 2024-02-09:
Hi, 

Is it possible to have the user_scn and update_hfield()  from passive_viewer in Renderer? I want the screens of my viewer and render synchronized so I can record what I see. As in the video below, the above two features of the viewer help me visualize my results well. Due to these custom artifacts, I can only record them from an external screen recorder, which is a hassle.  

https://github.com/google-deepmind/mujoco/assets/54435909/5b526f39-19f6-4a60-a026-9cfb5eba4741

saran-t, 2024-02-20:
While this isn't currently supported, the `Renderer` class is a pure Python helper. If you want, you should be able to fairly straightforwardly add this feature (and maybe consider sending it in as a PR!)

Dashsouradeep, 2025-12-25:
Hi! I would like to work on this as my first open-source contribution.

Before I start, could you please confirm if adding Renderer API equivalents for user_scn and update_hfield, along with Python bindings and tests, is the right approach for this feature?

Or is there any other idea to implement this?

Thanks!