google-deepmind/mujoco #1414: Issue with Box Rotation Around Z-axis and Movement in XYZ Plane

bheijden, 2024-02-12:
Hello,

I need assistance with configuring a scene where a box can rotate (yaw) around its z-axis and also move within the xyz plane. I've included a partial XML definition and a video demonstrating the current behavior (in MJX), where the box unexpectedly rotates around one of its corners rather than its center of mass. Can anyone identify what might be causing this issue?
Here is a model which explains my question:

<details>
  <summary>minimal XML</summary>

```XML
<mujoco>
<worldbody>
        <light cutoff="100" diffuse="1 1 1" dir="-0 0 -1.3" directional="true" exponent="1" pos="0 0 1.3" specular=".1 .1 .1"/>
        <!-- table -->
        <geom name="table" size="3 3 .2" type="plane" material="grid" contype="8" conaffinity="11" condim="3"/>
        <!-- box -->
        <body name="box" pos="0.0 0.0 0.0" >
            <geom name="box_geom" rgba="0.2 0.2 0.2 1" type="box" size="0.05 0.05 0.05" density="47" contype="7" conaffinity="8" condim="3"/>
            <joint name="box_slidex" type="slide" pos="0 0 0" axis="1 0 0" range="-2 2" damping="0."/>
            <joint name="box_slidey" type="slide" pos="0 0 0" axis="0 1 0" range="-2 2" damping="0."/>
            <joint name="box_slidez" type="slide" pos="0 0 0" axis="0 0 1" range="-2 0.3" damping="0."/>
            <joint name="box_yaw" type="hinge" pos="0 0 0" axis="0 0 1" limited="false" damping="0."/>
        </body>
  </worldbody>
</mujoco>
```

</details>

Here is a screenshot / video, illustrating my question:

https://github.com/google-deepmind/mujoco/assets/45761969/0b09574b-dabb-4080-97a5-c974db074616

yuvaltassa, 2024-02-12:
The model looks fine.

Does this happen in regular MuJoCo? It shouldn't.

@btaba @erikfrey correct me if I'm wrong but I think MJX only ever computes 1 contact point between pairs of geoms, even with planes. In vanilla MuJoCo you would get 4 contacts and the box wouldn't do this. If you had a free joint the contact point would jump around and prevent this, but as your box cannot roll or pitch, one corner is selected arbitrarily and the box rotates around it (so all 3 planar joints are moving, it's not the case that the joint is in the corner). [Or maybe there is a bug in the MJX box-plane collider 😛]

If my diagnosis is correct, your workaround for now if you want to use MJX, is to put 4 tiny spheres on the bottom corners.

btaba, 2024-02-12:
Hi @bheijden thanks for the bug report!

In principal, MJX gets 4 contact points for any convex<>primitive or convex<>convex collision. In practice, it looks like box-plane picks 1 degenerate point, so only 3 contact points are being active rather than on all four corners. As Yuval mentioned adding a sphere to each box corner would get the desired behavior, but we'll push a fix to pick all 4 points for box-plane. 

[image]

btaba, 2024-02-12:
The contact point part of the fix  https://github.com/google-deepmind/mujoco/commit/805d862d5feffe5a4449afc0605b990baf7aab07

btaba, 2024-02-13:
Ok I played with a version of the box example at HEAD, and it looks good to me.  @bheijden let us know if you still see weird behavior.

<details>
  <summary>XML</summary>

```XML
<mujoco>
<worldbody>
        <light cutoff="100" diffuse="1 1 1" dir="-0 0 -1.3" directional="true" exponent="1" pos="0 0 1.3" specular=".1 .1 .1"/>
        <!-- table -->
        <geom name="table" size="3 3 .2" type="plane" contype="8" conaffinity="11" condim="3"/>
        <!-- box -->
        <body name="box" pos="0.0 0.0 0.0" >
            <geom name="box_geom" rgba="0.2 0.2 0.2 1" type="box" size="0.05 0.05 0.05" density="47" contype="7" conaffinity="8" condim="3"/>
            <joint name="x" type="slide" pos="0 0 0" axis="1 0 0" range="-2 2" damping="0."  armature="0." stiffness="0." />
            <joint name="y" type="slide" pos="0 0 0" axis="0 1 0" range="-2 2" damping="0."  armature="0." stiffness="0." />
            <joint name="z" type="slide" pos="0 0 0" axis="0 0 1" range="-2 0.3" damping="0."  armature="0." stiffness="0." />
            <joint name="box_yaw" type="hinge" pos="0 0 0" axis="0 0 1" limited="false" damping="0."  armature="0." stiffness="0." />
        </body>
  </worldbody>
<actuator>
 <motor joint="x"/>
 <motor joint="y"/>
 <motor joint="z"/>
</actuator>
```
</details>

bheijden, 2024-02-14:
Thanks for the fix! Tried to test it out yesterday but had some trouble setting up Mujoco locally. I'll probably test it out then, and until then use the sphere fix!

btaba, 2024-03-11:
Any updates @bheijden ? Is this still an issue?

bheijden, 2024-03-13:
It's on my to do list to check out this weekend! Will keep you posted. I'll close this issue if it's fixed!

bheijden, 2024-03-18:
Did not find the time to test yet. Probably won't happen until after the next ~2 weeks (busy with paper rebuttal).... But, it's definitely something I'll get back to, so you can leave it open for now.

kevinzakka, 2024-08-15:
Going to close, feel free to re-open if the issue is unsolved.

closed 2024-08-15 by kevinzakka