google-deepmind/mujoco #1415: sample compiling cannot find framework on MacOS

Valeriyag, 2024-02-12:
Hi,

I'm an Associate Professor at West Virginia University and I'm trying to use MuJoCo for computational neuroscience. I have little experience in compiling. I am trying to figure out how to link MuJoCo to Matlab via Mex functions. I've started with compiling sample code that came with the MuJoCo download.

I'm looking for some help with compiling the code from Sample folder. I have copied the MuJoCo app into the Applications folder and it runs fine. Then in terminal I ran the Makefile and it causes an error that the framework mujoco is not found.

The code I run in terminal:
 cd /Users/vgritsenko/MuJoCo_sample
 make

The output I get:
clang++ -O2 -F/Applications/MuJoCo.app/Contents/Frameworks -I/usr/local/include -pthread -std=c++17 -stdlib=libc++ -L/usr/local/lib -Wl,-rpath,/Applications/MuJoCo.app/Contents/Frameworks    testspeed.cc  -framework mujoco        -o testspeed
ld: framework 'mujoco' not found
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make: *** [all] Error 1

When I go to the MuJoCo framework path, the framework is there:

[image]

Any advice is appreciated.

yuvaltassa, 2024-02-14:
Did you [this](https://github.com/google-deepmind/mujoco?tab=readme-ov-file#third-party-bindings)?

I recommend talking to @vmanoj1996.

Valeriyag, 2024-02-15:
Thanks, I've reached out to Manoj directly.

I have downloaded the simulink blockset and, unfortunately, I get the same error when trying to compile Mex functions from Matlab. This makes me think that there is something I'm missing in my MacOS Sonoma 14.2.1.

kathirgounder, 2024-07-11:
Hi was there ever a resolution for this, running into the same issue on MacOS