google-deepmind/mujoco #1424: problem with Colab's MJX Tutorial

mnm117, 2024-02-16:
Hi,

I'm a student and I'm trying to use MuJoCo for my research.

I'm looking for some help with running the tutorial on my PC.

Here is a model which explains my question:

<details>
When I wanted to run the MJX tutorial and running the sample on my pc, I faced error. the below is my code that I copied from colab to a python file on my PC (Ubuntu 22.04.03) 

```
#@title Check if MuJoCo installation was successful

import distutils.util
import os
import subprocess
if subprocess.run('nvidia-smi').returncode:
  raise RuntimeError(
      'Cannot communicate with GPU. '
      'Make sure you are using a GPU Colab runtime. '
      'Go to the Runtime menu and select Choose runtime type.')

# Add an ICD config so that glvnd can pick up the Nvidia EGL driver.
# This is usually installed as part of an Nvidia driver package, but the Colab
# kernel doesn't install its driver via APT, and as a result the ICD is missing.
# (https://github.com/NVIDIA/libglvnd/blob/master/src/EGL/icd_enumeration.md)
NVIDIA_ICD_CONFIG_PATH = '/usr/share/glvnd/egl_vendor.d/10_nvidia.json'
if not os.path.exists(NVIDIA_ICD_CONFIG_PATH):
  with open(NVIDIA_ICD_CONFIG_PATH, 'w') as f:
    f.write("""{
    "file_format_version" : "1.0.0",
    "ICD" : {
        "library_path" : "libEGL_nvidia.so.0"
    }
}
""")

# Tell XLA to use Triton GEMM, this improves steps/sec by ~30% on some GPUs
xla_flags = os.environ.get('XLA_FLAGS', '')
xla_flags += ' --xla_gpu_triton_gemm_any=True'
os.environ['XLA_FLAGS'] = xla_flags

# Configure MuJoCo to use the EGL rendering backend (requires GPU)
print('Setting environment variable to use GPU rendering:')
import os

# Set the MUJOCO_GL environment variable to 'egl'
os.environ['MUJOCO_GL'] = 'egl'

try:
  print('Checking that the installation succeeded:')
  import mujoco 
  mujoco.MjModel.from_xml_string('<mujoco/>')
except Exception as e:
  raise e from RuntimeError(
      'Something went wrong during installation. Check the shell output above '
      'for more information.\n'
      'If using a hosted Colab runtime, make sure you enable GPU acceleration '
      'by going to the Runtime menu and selecting "Choose runtime type".')

print('Installation successful.')

#@title Import packages for plotting and creating graphics
import time
import itertools
import numpy as np
from typing import Callable, NamedTuple, Optional, Union, List

# Graphics and plotting.
print('Installing mediapy:')

import mediapy as media
import matplotlib.pyplot as plt

# More legible printing from numpy.
np.set_printoptions(precision=3, suppress=True, linewidth=100)

#@title Import MuJoCo, MJX, and Brax

from datetime import datetime
import functools
from IPython.display import HTML
import jax
from jax import numpy as jp
import numpy as np
from typing import Any, Dict, Sequence, Tuple, Union

from brax import base
from brax import envs
from brax import math
from brax.base import Base, Motion, Transform
from brax.envs.base import Env, PipelineEnv, State
from brax.mjx.base import State as MjxState
from brax.training.agents.ppo import train as ppo
from brax.training.agents.ppo import networks as ppo_networks
from brax.io import html, mjcf, model

from etils import epath
from flax import struct
from matplotlib import pyplot as plt
import mediapy as media
from ml_collections import config_dict
import mujoco
from mujoco import mjx

xml = """
<mujoco>
  <worldbody>
    <light name="top" pos="0 0 1"/>
    <body name="box_and_sphere" euler="0 0 -30">
      <joint name="swing" type="hinge" axis="1 -1 0" pos="-.2 -.2 -.2"/>
      <geom name="red_box" type="box" size=".2 .2 .2" rgba="1 0 0 1"/>
      <geom name="green_sphere" pos=".2 .2 .2" size=".1" rgba="0 1 0 1"/>
    </body>
  </worldbody>
</mujoco>
"""

# Make model, data, and renderer
mj_model = mujoco.MjModel.from_xml_string(xml)
mj_data = mujoco.MjData(mj_model)
renderer = mujoco.Renderer(mj_model)
```

</details>
the error is :
    from gym.vector import utils
ModuleNotFoundError: No module named 'gym.vector'

I tried to check Mujoco installation, So I wrote " import mujoco_py" in python and I faced the below error.

/usr/lib/python3/dist-packages/pythran/tables.py:4520: FutureWarning: In the future `np.bool` will be defined as the corresponding NumPy scalar.
  if not hasattr(numpy, method):
/usr/lib/python3/dist-packages/pythran/tables.py:4553: FutureWarning: In the future `np.bytes` will be defined as the corresponding NumPy scalar.
  obj = getattr(themodule, elem)
Compiling /home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx because it changed.
[1/1] Cythonizing /home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx
performance hint: /home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:67:5: Exception check on 'c_warning_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_warning_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_warning_callback' to allow an error code to be returned.
performance hint: /home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:104:5: Exception check on 'c_error_callback' will always require the GIL to be acquired.
Possible solutions:
	1. Declare 'c_error_callback' as 'noexcept' if you control the definition and you're sure you don't want the function to raise exceptions.
	2. Use an 'int' return type on 'c_error_callback' to allow an error code to be returned.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_warning_callback
    global mju_user_warning
    py_warning_callback = warn
    mju_user_warning = c_warning_callback
                       ^
------------------------------------------------------------

/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:92:23: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const char *) except * nogil'.

Error compiling Cython file:
------------------------------------------------------------
...
    See c_warning_callback, which is the C wrapper to the user defined function
    '''
    global py_error_callback
    global mju_user_error
    py_error_callback = err_callback
    mju_user_error = c_error_callback
                     ^
------------------------------------------------------------

/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx:127:21: Cannot assign type 'void (const char *) except * nogil' to 'void (*)(const char *) noexcept nogil'. Exception values are incompatible. Suggest adding 'noexcept' to type 'void (const char *) except * nogil'.
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/__init__.py", line 2, in <module>
    from mujoco_py.builder import cymj, ignore_mujoco_warnings, functions, MujocoException
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 504, in <module>
    cymj = load_cython_ext(mujoco_path)
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 110, in load_cython_ext
    cext_so_path = builder.build()
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 226, in build
    built_so_file_path = self._build_impl()
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 296, in _build_impl
    so_file_path = super()._build_impl()
  File "/home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/builder.py", line 239, in _build_impl
    dist.ext_modules = cythonize([self.extension])
  File "/home/mohsen/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1154, in cythonize
    cythonize_one(*args)
  File "/home/mohsen/.local/lib/python3.10/site-packages/Cython/Build/Dependencies.py", line 1321, in cythonize_one
    raise CompileError(None, pyx_file)
Cython.Compiler.Errors.CompileError: /home/mohsen/.local/lib/python3.10/site-packages/mujoco_py/cymj.pyx

Can anyone help me?

btaba, 2024-02-27:
Hi @mnm117 , thanks for the bug report.  `gym` is a brax [dependency](https://github.com/google/brax/blob/3c109cfd131e01691a53891e8f2ec9b32cf97670/setup.py#L49)

How did you install `brax`?

mnm117, 2024-02-28:
Hello @btaba 
Thank you for your response.
I install it with
`pip install brax`

btaba, 2024-02-28:
Hi @mnm117 , can you send the traceback for the pip install for brax and the full traceback for "ModuleNotFoundError: No module named 'gym.vector'".   You should have gym installed

btaba, 2024-03-11:
Closing due to inactivity, please open a new issue with more details on how to reproduce the error

closed 2024-03-11 by btaba

IndefiniteCalculus, 2024-05-31:
I has same issue when I try this tutorial [https://www.roboti.us/download.html](url). I install gym and manually download zip from [https://www.roboti.us/download.html](url). I use python 3.11.9 and `pip install cython==3.0.10`. The output is
`---------------------------------------------------------------------------
CompileError                              Traceback (most recent call last)
Cell In[8], [line 51](vscode-notebook-cell:?execution_count=8&line=51)
     [48](vscode-notebook-cell:?execution_count=8&line=48)         pass
     [50](vscode-notebook-cell:?execution_count=8&line=50) # Create and wrap the environment
---> [51](vscode-notebook-cell:?execution_count=8&line=51) env = gym.make("InvertedPendulum-v4")
     [52](vscode-notebook-cell:?execution_count=8&line=52) wrapped_env = gym.wrappers.RecordEpisodeStatistics(env, 50)  # Records episode-reward
     [54](vscode-notebook-cell:?execution_count=8&line=54) total_num_episodes = int(5e3)  # Total number of episodes

File d:\Compiler\Python\Anaconda\anaconda3\envs\openAi-rl\Lib\site-packages\gymnasium\envs\registration.py:756, in make(id, max_episode_steps, autoreset, apply_api_compatibility, disable_env_checker, **kwargs)
    [753](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:753)     env_creator = env_spec.entry_point
    [754](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:754) else:
    [755](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:755)     # Assume it's a string
--> [756](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:756)     env_creator = load_env_creator(env_spec.entry_point)
    [758](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:758) # Determine if to use the rendering
    [759](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:759) render_modes: list[str] | None = None

File d:\Compiler\Python\Anaconda\anaconda3\envs\openAi-rl\Lib\site-packages\gymnasium\envs\registration.py:545, in load_env_creator(name)
    [536](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:536) """Loads an environment with name of style ``"(import path):(environment name)"`` and returns the environment creation function, normally the environment class type.
    [537](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:537) 
    [538](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:538) Args:
   (...)
    [542](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:542)     The environment constructor for the given environment name.
    [543](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:543) """
    [544](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:544) mod_name, attr_name = name.split(":")
--> [545](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:545) mod = importlib.import_module(mod_name)
    [546](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:546) fn = getattr(mod, attr_name)
    [547](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/registration.py:547) return fn

File d:\Compiler\Python\Anaconda\anaconda3\envs\openAi-rl\Lib\importlib\__init__.py:126, in import_module(name, package)
    [124](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/importlib/__init__.py:124)             break
    [125](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/importlib/__init__.py:125)         level += 1
--> [126](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/importlib/__init__.py:126) return _bootstrap._gcd_import(name[level:], package, level)

File <frozen importlib._bootstrap>:1204, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1176, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:1126, in _find_and_load_unlocked(name, import_)

File <frozen importlib._bootstrap>:241, in _call_with_frames_removed(f, *args, **kwds)

File <frozen importlib._bootstrap>:1204, in _gcd_import(name, package, level)

File <frozen importlib._bootstrap>:1176, in _find_and_load(name, import_)

File <frozen importlib._bootstrap>:1147, in _find_and_load_unlocked(name, import_)

File <frozen importlib._bootstrap>:690, in _load_unlocked(spec)

File <frozen importlib._bootstrap_external>:940, in exec_module(self, module)

File <frozen importlib._bootstrap>:241, in _call_with_frames_removed(f, *args, **kwds)

File d:\Compiler\Python\Anaconda\anaconda3\envs\openAi-rl\Lib\site-packages\gymnasium\envs\mujoco\__init__.py:[1](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/gymnasium/envs/mujoco/__init__.py:1)
----> 1 from gymnasium.envs.mujoco.mujoco_env import MujocoEnv, MuJocoPyEnv  # isort:skip
[... cut ...]
-> [1321](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/Cython/Build/Dependencies.py:1321)         raise CompileError(None, pyx_file)
   [1322](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/Cython/Build/Dependencies.py:1322)     elif os.path.exists(c_file):
   [1323](file:///D:/Compiler/Python/Anaconda/anaconda3/envs/openAi-rl/Lib/site-packages/Cython/Build/Dependencies.py:1323)         os.remove(c_file)

CompileError: d:\Compiler\Python\Anaconda\anaconda3\envs\openAi-rl\Lib\site-packages\mujoco_py\cymj.pyx`