google-deepmind/mujoco #1427: Running mujoco.viewer in a Wayland session results in borderless window

zmk5, 2024-02-17:
Running the `mujoco.viewer` python binding results in a borderless window. The program will continue to work normally, but cannot be closed except by a keyboard interrupt. I have attached an image showing how the windowed screen looks like:

[image]

Running the command also results in the following warnings:

```bash
~$ python3 -m mujoco.viewer
/home/zmk5/.local/lib/python3.10/site-packages/glfw/__init__.py:916: GLFWError: (65544) b'Wayland: Window position retrieval not supported'
  warnings.warn(message, GLFWError)
/home/zmk5/.local/lib/python3.10/site-packages/glfw/__init__.py:916: GLFWError: (65544) b'Wayland: Standard cursor shape unavailable'
  warnings.warn(message, GLFWError)
```

I've tested this on sample code located [here](https://github.com/kevinzakka/mjctrl/blob/main/diffik.py) and results in the same borderless window and warnings.

- Include the following context:
  - Operating system: Ubuntu 22.04.3 (amd64)
  - MuJoCo version: 3.1.1
  - For Python issues, what bindings are you using (i.e `mujoco`, `dm_control`, `mujoco-py`)?
    - This is with the `mujoco` Python bindings.

saran-t, 2024-02-20:
Does this also affect the `simulate` program? (You can just download our prebuilt tarball and execute `simulate`.)

zmk5, 2024-02-23:
Hi @saran-t , I just checked and it doesn't seem to affect the `simulate` program in the prebuilt tarball.

[image]

saran-t, 2024-02-23:
That is very strange... the two viewers share the same code.

You aren't building your own Python bindings right?

zmk5, 2024-03-05:
No I'm not. This is just from running `pip install mujoco` on a default Ubuntu 22.04 install running Wayland.

saran-t, 2024-03-05:
OK, this is most likely because the Python bindings depend on GLFW via the `PyGLFW` package, while `simulate` gets it from the system.

Could you please try editing [this file](https://github.com/google-deepmind/mujoco/blob/main/python/mujoco/viewer.py) as follows:
- Comment out `import glfw` at line 29
- Comment out lines 34-36
- Unindent line 37 and change function argument to `ctypes.CDLL(ctypes.util.find_library('glfw'))._handle`

Let me know if this helps.

zmk5, 2024-03-07:
Thank you for the suggestion! I will try it and see if it works.

nsk126, 2024-03-16:
Were you able to solve this problem? I have a similar issue where I'm unable to interact with Viewer window. I also have Wayland.

```bash
/home/user/anaconda3/envs/my_env/lib/python3.9/site-packages/glfw/__init__.py:914: GLFWError: (65544) b'Wayland: Window position retrieval not supported'
  warnings.warn(message, GLFWError)
WARNING:absl:OpenGL error 0x502 in or before mjr_makeContext
```

I'm using:
```
python==3.9
mujoco==3.1.3
```

saran-t, 2024-03-17:
We depend on GLFW via PyGLFW since I assumed that Python users may be using PyGLFW elsewhere in their codebase, and pulling in two different copies of GLFW (one from the system and one from PyGLFW) may cause symbol conflicts.

However, it looks like PyGLFW doesn't build with Wayland support enabled. The suggested workaround skips PyGLFW and loads GLFW from the system.

A more permanent solution would either be to detect whether the user is running in Wayland and use the workaround if they are, or request that PyGLFW enables Wayland support in their next version.

zmk5, 2024-03-17:
According to [this](https://github.com/FlorianRhiem/pyGLFW?tab=readme-ov-file#linux), PyGLFW attempts to detect which display server the user is using, but the option can be forced using the `PYGLFW_LIBRARY_VARIANT` environment variable to be `wayland`.

```bash
export PYGLFW_LIBRARY_VARIANT=wayland
```

I will need to test this if it works.

lieskjur, 2024-03-20:
I have the same issue as @nsk126.  I also tried @saran-t  solution. After adding `import ctypes` I am now getting `cannot dlsym glfwCreateWindow` any ideas on how to proceed?

nsk126, 2024-03-23:
I think setting the glfw handle dynamically should be enough to handle this issue?

something like this in line `35-38`
```python
if not glfw._glfw:  # pylint: disable=protected-access
  raise RuntimeError('GLFW dynamic library handle is not available')
else:
  _simulate.set_glfw_dlhandle(ctypes.CDLL(ctypes.util.find_library('glfw'))._handle)  # pylint: disable=protected-access
```

i reverted to a older version of my code with just this implementation and it seems to work fine.

@lieskjur you'll have to explain a bit more detail on where you're getting the error from.

adlarkin, 2024-03-27:
@nsk126 what do you mean by "an older version of my code?" Can you share which versions of MuJoCo and glfw you're using?

nsk126, 2024-03-28:
By reverting to my old code, I meant I just turned line 37 in mujoco.viewer into 

```python
 _simulate.set_glfw_dlhandle(ctypes.CDLL(ctypes.util.find_library('glfw'))._handle)
```

Everything else can stay the same.

adlarkin, 2024-03-30:
That doesn't seem to work for me when running wayland on Ubuntu 22.04, but I can revert to using x11 for now. When I use x11, I don't have to change anything in the mujoco pip installation.

darkopetrovic, 2024-04-01:
I am working in WSL in a docker container and had the same issue. 

The problem was the version of MESA driver. It seems that the version must at least > 21.x.

Furthermore, by updating the driver my GPU card was successfully detected. 

Below the output of `glxinfo`. 

```
$ glxinfo -B
name of display: :0
display: :0  screen: 0
direct rendering: Yes
Extended renderer info (GLX_MESA_query_renderer):
    Vendor: Microsoft Corporation (0xffffffff)
    Device: D3D12 (NVIDIA GeForce GTX 1660) (0xffffffff)
    Version: 23.1.2
    Accelerated: yes
    Video memory: 22321MB
    Unified memory: no
    Preferred profile: core (0x1)
    Max core profile version: 4.2
    Max compat profile version: 4.2
    Max GLES1 profile version: 1.1
    Max GLES[23] profile version: 3.1
OpenGL vendor string: Microsoft Corporation
OpenGL renderer string: D3D12 (NVIDIA GeForce GTX 1660)
OpenGL core profile version string: 4.2 (Core Profile) Mesa 23.1.2-1~mx23ahs
OpenGL core profile shading language version string: 4.20
OpenGL core profile context flags: (none)
OpenGL core profile profile mask: core profile

OpenGL version string: 4.2 (Compatibility Profile) Mesa 23.1.2-1~mx23ahs
OpenGL shading language version string: 4.20
OpenGL context flags: (none)
OpenGL profile mask: compatibility profile

OpenGL ES profile version string: OpenGL ES 3.1 Mesa 23.1.2-1~mx23ahs
OpenGL ES profile shading language version string: OpenGL ES GLSL ES 3.10
```

lounick, 2024-05-07:
@adlarkin the `cannot dlsym glfwCreateWindow`  is caused by not being able to find `libglfw.so` in any location. If you set the `LD_LIBRARY_PATH` to include the environment directory that pulls the library it should find it. For example in my case `export LD_LIBRARY_PATH="/home/foo/git/bar/env/lib/python3.12/site-packages/glfw/wayland/"` is something like this.

However @saran-t even if it is found the window remains unresponsive and have to kill the process. Using the upstream provided GLFW (I'm on Arch) it works (i.e. is not frozen, buttons click etc.). So I would say the problem is the library pulled by pyGLFW. The best would be for them to build correctly with Wayland?

If there is anything else I can test please let me know.

owenchen0408, 2024-07-10:
@saran-t I tried your solution but it says
 `cannot dlsym glfwCreateWindow
Aborted (core dumped)`
Is there any ned solutions to this? thanks!

Troldmand43, 2024-09-20:
@zmk5 I had the same warnings, and my simulation was just frozen. I fixed the error by using Xorg instead of Wayland. 
https://linuxconfig.org/how-to-enable-disable-wayland-on-ubuntu-22-04-desktop

QhuangL, 2024-09-20:
> @zmk5 I had the same warnings, and my simulation was just frozen. I fixed the error by using Xorg instead of Wayland. https://linuxconfig.org/how-to-enable-disable-wayland-on-ubuntu-22-04-desktop

Solve my problem, thanks a lot

saran-t, 2025-07-14:
@m01 could you maybe try to repro this?

yuvaltassa, 2025-09-01:
@m01 is this now fixed?

erikfrey, 2026-09-14:
For what it's worth, `export PYGLFW_LIBRARY_VARIANT=x11` fixed this for me (Ubuntu 26.04, GNOME/Wayland, glfw 2.10.0 with bundled GLFW 3.4.0). Forces the X11-capable build through XWayland, sidestepping the Wayland build's libdecor dependency entirely.