google-deepmind/mujoco #1432: Support loading mesh specified by package:// URIs when loading URDF models

traversaro, 2024-02-20:
**Is your feature request related to a problem? Please describe.**

A common problem we are experiencing with users using MuJoCo with iCub models (see https://github.com/robotology/icub-models/issues/155 for a related issue) are related to the fact that the MuJoCo URDF parser strips any path information from the `filename` attribute of meshes. See https://github.com/google-deepmind/mujoco/blob/7c534a066c8d61bf87f7818624baf6fbe500c17b/src/xml/xml_urdf.cc#L565-L572 for the related snippet code. It would be great if MuJoCo had at least an option to avoid this behaviour, and instead actually resolve URDF's package:// URIs. 

**Describe the solution you'd like**

Possible solutions include look for meshes specified via `package://` URIs by inspecting the values of `ROS_PACKAGE_PATH` (for ROS1-style packages) or `AMENT_PREFIX_PATH` (for ROS2-style packages). This can be done without any specific dependency on ROS via some dedicated code. For an example of how this is implemented in other libraries that consume URDFs like iDynTree or  Pinocchio see the following snippets:
* iDynTree: https://github.com/robotology/idyntree/blob/ff407448de6fbc635353dbdb31dcc3d913f15820/src/model/src/SolidShapes.cpp#L216-L308 
* pinocchio: https://github.com/stack-of-tasks/pinocchio/blob/0caf0ca4d07e63834cdc420c703993662c59e01b/src/parsers/urdf/geometry.cpp#L460-L462 and https://github.com/stack-of-tasks/pinocchio/blob/0caf0ca4d07e63834cdc420c703993662c59e01b/src/utils/file-explorer.cpp#L55

For reference, a similar logic is implemented in the following minimal Python package https://github.com/ami-iit/resolve-robotics-uri-py/blob/main/src/resolve_robotics_uri_py/resolve_robotics_uri_py.py#L19-L68 . This code is not directly usable in the mujoco C++ URDF parser, but it just meant to show how the logic works. 

**Describe alternatives you've considered**

An alternative choice done by Drake, is instead not to hardcode the lookup for any environment variable inside the library itself, but rather expose an API for users to specify the directories in which to look for files specified via `package://` URIs, see https://github.com/RobotLocomotion/drake-external-examples/issues/170 for an example of this. While this may require more work on the users to actually successfully parse URDF files, that would also a way to fix the problem.

**Additional context**

I will be happy to eventually work on this, but I would like to understand if maintainers are interested in this.

traversaro, 2024-02-20:
fyi @CarlottaSartore @giotherobot @siddharthdeore @diegoferigo @flferretti @akhilsathuluri

saran-t, 2024-02-20:
This should be done as a [resource provider](https://mujoco.readthedocs.io/en/stable/programming/extension.html#resource-providers).

traversaro, 2024-02-24:
> This should be done as a [resource provider](https://mujoco.readthedocs.io/en/stable/programming/extension.html#resource-providers).

Thanks, I built a (quick and dirty) prototype using the resource provider API, available at https://github.com/traversaro/mujoco-urdf-package-uri . 

I tried to build it as a project external to mujoco, so that I could use it also with existing version of MuJoCo, but unfortunately I had to modify MuJoCo itself due to the code in https://github.com/google-deepmind/mujoco/blob/7c534a066c8d61bf87f7818624baf6fbe500c17b/src/xml/xml_urdf.cc#L565-L572 that was always stripping all the path and the extension from the meshes. So I did the modifications in https://github.com/traversaro/mujoco/commit/d3d1ff019e967dbe56f028a7884a4d32a22d053a . 

In the example, then I am able to load unmodified URDF files for Panda and iCub:

[image]

[image]

At this point, probably we need to understand if you are interested in having such as resource provider inside MuJoCo itself, or if you imagine this as something that would live outside of mujoco.

akhilsathuluri, 2024-02-26:
One place where I see the utility of a `package_path` resolver is when we deal with predefined meshes (as modules) that are assembled as a part of a larger simulation. For example, when using a library like [Timor](https://gitlab.lrz.de/tum-cps/timor-python) or [DoD](https://github.com/akhilsathuluri/automatic-design-on-demand), where one ships module meshes along with the library that a user can assemble. In this case, auto-resolving paths is helpful, keeping the generated URDFs, any additional external meshes used, and the pre-defined meshes separate. That being said, Workflow2 works fine in this case too.

traversaro, 2024-02-26:
> One place where I see the utility of a `package_path` resolver is when we deal with predefined meshes (as modules) that are assembled as a part of a larger simulation. For example, when using a [...]

Indeed this is why I was discussing about the fact that you can't "include" a URDF in a bigger file in MuJoCo in https://github.com/google-deepmind/mujoco/issues/1432#issuecomment-1962918624 . Without the "include" feature, I guess there is less need for automatically resolving `package:/` URIs, right

akhilsathuluri, 2024-02-26:
> Indeed this is why I was discussing about the fact that you can't "include" a URDF in a bigger file in MuJoCo in [#1432 (comment)](https://github.com/google-deepmind/mujoco/issues/1432#issuecomment-1962918624) . Without the "include" feature, I guess there is less need for automatically resolving `package:/` URIs, right

That is correct. If you want to "include" a URDF into a bigger file, then it is perhaps not so useful. But what I meant, is to use raw meshes and create a URDF on the fly, for example via [odio_urdf](https://github.com/hauptmech/odio_urdf).

yuvaltassa, 2026-09-08:
This is very old now. I still think this is a nice thing though, Please reopen a new PR if you want it

closed 2026-09-08 by yuvaltassa