google-deepmind/mujoco #1433: Shirt/Cloth Simulation with `<flexcomp>` diverging

thisjustin123, 2024-02-20:
Hi!

I'm an undergrad student trying to use MuJoCo for cloth simulation, specifically that of a t-shirt, for robot-assisted dressing.

I'm looking for some help with setting up the XML file correctly with the `elasticity` plugin and other `flexcomp` parameters to get the shirt to behave correctly.

Here is the shirt XML I'm currently working with:

<details>
  <summary>Shirt `worldbody` XML</summary>

```XML
<worldbody>
    <geom name="floor" pos="0 0 -1" size="0 0 .25" type="plane" material="plane" condim="3" />

    <!-- Shirt -->
    <flexcomp name="shirt_flex" type="mesh" radius="0.01" dim="2" pos="0 0 2"
      file="shirt_lowpoly_tetra_again.obj">
      <contact solref="0.003" />
      <edge equality="true" damping="0.1" />
      <plugin plugin="mujoco.elasticity.shell">
        <config key="poisson" value="0" />
        <config key="thickness" value="8e-3" />
        <config key="young" value="3e5" />
      </plugin>
    </flexcomp>
</worldbody>
```

</details>

When I run the simulation with `simulate.exe`, I see the shirt violently diverge to infinity and crash repeatedly:

https://github.com/google-deepmind/mujoco/assets/47724806/5038c8d8-a199-43ae-ab69-3772f0ac79fb

I receive the following output in the simulate terminal:

[image]

Any idea on what I can change? I'm expecting similar-ish results to the `poncho_flex.xml` example provided with MuJoCo, which is what I used to model this scene after.

mohammad200h, 2024-02-20:
Hey, can you share shirt_lowpoly_tetra_again.obj with us? 

Here are some steps you can try on your own:
1.Ftetwild outputs .msh format. This format will be in gmsh2.2 

2. You need to convert this using gmsh app. Import the file with extention .msh generated by ftetwild and then export it to format 4.

3. Use the file exported by gmsh app in your xml. The file should have extension .msh.

If you share the .obj file I can also test it :)

quagla, 2024-02-21:
Does it still diverge if you reduce the time step?

thisjustin123, 2024-02-23:
> Hey, can you share shirt_lowpoly_tetra_again.obj with us?
>
> Here are some steps you can try on your own: 1.Ftetwild outputs .msh format. This format will be in gmsh2.2
>
> 2. You need to convert this [...]

Sure, here's the file I'm using: [Google Drive Link](https://drive.google.com/file/d/11pUYs6_11xArARo6Vsrve2lQddeM96zM/view?usp=sharing)

Although I did get a `.msh` mesh, I needed to use the `.stl` output since the `.stl` "tracked surface" mesh is the only once that has a hollow inside--the other mesh outputs all had closed sleeves, torso holes, etc. To generate my `.obj` file, I then converted this `.stl` file into a `.obj` with MeshLab.

I just tried converted the `.stl` to a `.msh` with gmsh, but get an error when using this mesh for my MJCF:

[image]

thisjustin123, 2024-02-23:
> Does it still diverge if you reduce the time step?

Unfortunately, yes. I reduced the timestep at least to where I can see the diverging happening incrementally, though, if it's helpful (reset happens at ~3 seconds):

https://github.com/google-deepmind/mujoco/assets/47724806/0e3d6141-0b47-4e8a-80a2-bad5beb636df

quagla, 2024-03-04:
Can you share the entire XML?

thisjustin123, 2024-03-04:
> Can you share the entire XML?

Of course! Here's the entire XML content (in `shirt.xml`). The timestep might be absurdly low just because I wanted to test the above behavior.

```xml
<mujoco model="Shirt">
  <option timestep="0.0002" solver="CG" iterations="30" tolerance="1e-6" />

  <size memory="200M" />

  <extension>
    <plugin plugin="mujoco.elasticity.shell" />
  </extension>

  <visual>
    <map force="0.1" zfar="30" />
    <rgba haze="0.15 0.25 0.35 1" />
    <quality shadowsize="2048" />
    <global offwidth="800" offheight="800" />
  </visual>

  <asset>
    <texture name="plane" type="2d" builtin="checker" rgb1=".2 .3 .4" rgb2=".1 0.15 0.2"
      width="512" height="512" mark="cross" markrgb=".8 .8 .8" />
    <texture name="hammock" type="2d" builtin="checker" rgb1=".1 .5 .1" rgb2=".5 .1 .1"
      width="512" height="512" mark="edge" markrgb=".8 .8 .8" />
    <material name="plane" reflectance="0.3" texture="plane" texrepeat="1 1" texuniform="true" />
    <material name="hammock" texture="hammock" />
  </asset>

  <worldbody>
    <geom name="floor" pos="0 0 -1" size="0 0 .25" type="plane" material="plane" condim="3" />

    <!-- Shirt -->
    <flexcomp name="shirt_flex" type="mesh" radius="0.01" dim="2" pos="0 0 2"
      file="shirt_lowpoly_tetra_again.obj">
      <contact solref="0.003" selfcollide="none" />
      <edge equality="true" damping="0.1" />
      <plugin plugin="mujoco.elasticity.shell">
        <config key="poisson" value="0" />
        <config key="thickness" value="8e-3" />
        <config key="young" value="3e5" />
      </plugin>
    </flexcomp>

    <!-- Lights -->
    <light directional="true" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1"
      castshadow="false" />
    <light directional="false" diffuse=".8 .8 .8" specular="0.3 0.3 0.3" pos="0 0 4" dir="0 0 -1" />
  </worldbody>
</mujoco>
```

Let me know if you'll need anything else!

quagla, 2024-03-06:
It looks like a time step stability issue to me. 

Also note that the cloth model assumes a flat equilibrium configuration. I haven't had the time yet to implement the case of a curved cloth. You should still get some smoothing effect by setting a lower bending that the one you have because anyway the bending forces are not strong enough to counteract gravity.

Finally, please check carefully that your mesh doesn't have very tiny triangles with nearly zero area.

quagla, 2024-03-06:
Hi @thisjustin123 I've been playing with the model and I can give the following recommendations:

- Use internal="false" in your flex contact parameters since there are vertices that overlap and generate huge contact forces (turn on `Contact Point` and `Flex Vert` in the rendering panel if you want to see).
- Decrease the `young` to something like 5e3, if I turn off gravity a value higher than this does not seem realistic for a cloth because it flattens everything too quickly.
- Perhaps increase the `damping` to 1 or 10.

closed 2024-03-06 by quagla