google-deepmind/mujoco #1435: VR setup instructions with Mujoco

eklavya1983, 2024-02-21:
Hello,

I am a beginner in using Mujoco.  I went through the tutorial and was able to run some basic simulations.  I am interested in learning more about how to enable VR integration for rending with mujoco.  In the documentation it is mentioned a code sample may be posted in the future.  I was wondering if someone can outline the steps I can follow for doing this.  From what I understand we need to render the views for left eye and right eye.  I would like to understand how we 
1. Specify the camera information for left eye, right eye
2. How we setup the openGL contexts
3. How we render

Thanks.

yuvaltassa, 2024-02-21:
We have an advanced draft of the VR utility, updated to latest MuJoCo. It should be submitted soon.

closed 2024-02-21 by yuvaltassa

eklavya1983, 2024-02-21:
Thank you!  Looking forward to it!

SkytAsul, 2024-07-02:
Hello! Do you believe it will be released soon?

yuvaltassa, 2024-07-02:
Please can check out #1452 and comment on the PR with any questions.
This is still under development but the basic functionality is there.

SkytAsul, 2024-07-31:
I've succeeded in making a VR app using MuJoCo. For future reference, I have written a paper [here](https://github.com/SkytAsul/DeformableSimulation/blob/main/mujocoxr-paper/MuJoCoXR.pdf) than contains a lot of useful information for future folks wanting to make a VR x MuJoCo software.
I hope it will benefit to some people!

yuvaltassa, 2024-07-31:
Amazing! cc @nishbo