google-deepmind/mujoco #1440: Composite/cable: vertex parsing error

danieldugas, 2024-02-24:
### First, a working case:

`mujoco-3.1.2/bin/simulate test_vertex_cable.xml`

test_vertex_cable.xml:
```
<mujoco model="coil">
   <extension>
      <plugin plugin="mujoco.elasticity.cable"/>
   </extension>

   <worldbody>

     <composite prefix="actuated1" type="cable"
       vertex="
       0 0 0
       1 0 0
       1 1 0
       0 1 0
       "
               offset="0 0 0" initial="free">
         <plugin plugin="mujoco.elasticity.cable">
            <!--Units are in Pa (SI)-->
            <config key="twist" value="5e8"/>
            <config key="bend" value="15e8"/>
            <config key="vmax" value="0"/>
         </plugin>
         <joint kind="main" damping="0.15" armature="0.01"/>
         <geom type="capsule" size=".005" rgba=".8 .2 .1 1"/>
      </composite>

   </worldbody>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 3.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
  </worldbody>
</mujoco>
```

[image]

### Example error case

`mujoco-3.1.2/bin/simulate test_vertex_cable_2.xml`

test_vertex_cable_2.xml:
```
<mujoco model="coil">
   <extension>
      <plugin plugin="mujoco.elasticity.cable"/>
   </extension>

   <worldbody>

     <composite prefix="actuated1" type="cable"
       vertex="
       0 0 0
       2 0 0
       2 1 0
       0 1 0
       "
               offset="0 0 0" initial="free">
         <plugin plugin="mujoco.elasticity.cable">
            <!--Units are in Pa (SI)-->
            <config key="twist" value="5e8"/>
            <config key="bend" value="15e8"/>
            <config key="vmax" value="0"/>
         </plugin>
         <joint kind="main" damping="0.15" armature="0.01"/>
         <geom type="capsule" size=".005" rgba=".8 .2 .1 1"/>
      </composite>

   </worldbody>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
      markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <worldbody>
    <light pos="0 0 3.5" dir="0 0 -1" directional="true"/>
    <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
  </worldbody>
</mujoco>
```

[image]

With vertices at [0, 0, 0] [2, 0, 0], [2, 1, 0], [0, 1, 0], we would expect a shape like this instead:

```
--------
       | 
--------
```

Here, 
Context:
  - Ubuntu 20.04
  - MuJoCo release 3.1.2

closed 2024-02-27 by copybara-service[bot]

quagla, 2024-02-27:
Thanks for flagging this! I fixed it. As a workaround before the next release (if you are not building from source), you can load and save the xml and edit the body positions by hand if you wish.

danieldugas, 2024-02-27:
That was amazingly fast, thanks!

commit 478ce0c8 (2024-02-27): Use previous edge length when setting the position of a cable body.

Fixes #1440.

PiperOrigin-RevId: 610690343
Change-Id: I0d5288e68069d1117c0c674530e0644960bab06c