google-deepmind/mujoco #1442: [MJX] Model loading errors in `brax.io`, for features supported by MJX

varadVaidya, 2024-02-25:
With the MuJoCo 3.1.2 update, the site transmission was added to MJX, and a new pipeline to train RL agents using `PipelineEnv` was added brax. However while loading the model, through this pipeline, errors regarding `RK4` integration scheme, and site transmission, are thrown, even though the pipeline is set to `mjx`. 

- Include the following context:
  - Operating system : Ubuntu 20.04
  - MuJoCo version (and if the bug is new, the version where it used to work): 3.1.2
  - mujoco python bindings

btaba, 2024-02-27:
Hi @varadVaidya thanks for the bug report. Can you please provide a way to reproduce the issue?

varadVaidya, 2024-02-27:
Oh yes, forgot to add the minimal code to reproduce the error. The python script is based on the MJX colab tutorial.
<details>
<summary>XML File</summary>
<br>

```XML
<mujoco model="tmp_box">
    <compiler inertiafromgeom="true" coordinate="local"/>
    <option integrator="RK4"/>
    <statistic center="0 0 0.1" extent="0.6" meansize=".05"/>
    <visual>
        <headlight diffuse="0.6 0.6 0.6" ambient="0.3 0.3 0.3" specular="0 0 0"/>
        <rgba haze="0.15 0.25 0.35 1"/>
        <global azimuth="-20" elevation="-20" ellipsoidinertia="true" offwidth="1920" offheight="1080"/>
    </visual>
    <asset>
        <!-- <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/> -->
        <texture name="skybox" type="skybox" builtin="gradient" rgb1=".4 .6 .8" rgb2="0 0 0" width="800" height="800"
            mark="random" markrgb="1 1 1"/>
        <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3"
            markrgb="0.8 0.8 0.8" width="300" height="300"/>
        <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
    </asset>

    <worldbody>
        <light name="spotlight" mode="targetbodycom" target="core" pos="0 -1 2"/>
        <geom name="floor" size="0 0 0.05" type="plane" material="groundplane"/>
        <body name="core" pos="0 0 0.4" quat="1 0 0 0">
            <joint name="root" type="free" damping="0" armature="0" pos="0 0 0" limited="false"/>

            <geom name="core_geom" type="box" pos="0 0 0" quat="1 0 0 0" size=".05 .05 .05" rgba=".3 .3 .8 1" mass=".1"/>
            <site name="thrust_com" type="cylinder" pos="0 0 0" quat="1 0 0 0"/>
        </body>
        <camera name="fixed" pos="-2.567 -2.042 2.828" xyaxes="0.623 -0.783 0.000 0.500 0.398 0.769"
            resolution="1920 1080"/>
    </worldbody>
    <actuator>
        <motor ctrllimited="true" ctrlrange="0 10" gear="0 0 1 0 0 0" site="thrust_com" name="body_thrust"/>
        <motor ctrllimited="true" ctrlrange="-0.01 0.01" gear="0 0 0 -1 0 0" site="thrust_com" name="x_moment"/>
        <motor ctrllimited="true" ctrlrange="-0.01 0.01" gear="0 0 0 0 -1 0" site="thrust_com" name="y_moment"/>
        <motor ctrllimited="true" ctrlrange="-0.01 0.01" gear="0 0 0 0 0 -1" site="thrust_com" name="z_moment"/>
    </actuator>

</mujoco>
```

</details>

<details>
<summary>Python Code</summary>
<br>

```python
import os
# set MUJOCO_GL=egl as environment variable
os.environ["MUJOCO_GL"] = "egl"
os.environ["XLA_PYTHON_CLIENT_MEM_FRACTION"] = "0.4"

import time
import itertools
import numpy as np
np.set_printoptions(precision=3, suppress=True, linewidth=100)
from typing import Callable, NamedTuple, Optional, Union, List
import matplotlib.pyplot as plt

from datetime import datetime
import functools
from IPython.display import HTML
import jax
from jax import numpy as jp
[... 112 lines cut ...]
    return _envs[env_name](**kwargs)
  File "/home/varad/study/robotics/mujoco_sims/mjx_test/server/test_box.py", line 62, in __init__
    sys = mjcf.load_model(model)
  File "/home/varad/venv/mjx/lib/python3.10/site-packages/brax/io/mjcf.py", line 260, in load_model
    raise NotImplementedError('Only euler integration is supported.')
NotImplementedError: Only euler integration is supported.
```
Removing the `RK4` from XML shows:
```
An NVIDIA GPU may be present on this machine, but a CUDA-enabled jaxlib is not installed. Falling back to cpu.
[0.  0.  0.4 1.  0.  0.  0. ] <class 'numpy.ndarray'>
[0.  0.  0.4 1.  0.  0.  0. ] <class 'jaxlib.xla_extension.ArrayImpl'> {CpuDevice(id=0)}
Traceback (most recent call last):
  File "/home/varad/study/robotics/mujoco_sims/mjx_test/server/test_box.py", line 95, in <module>
    env = envs.get_environment(env_name)
  File "/home/varad/venv/mjx/lib/python3.10/site-packages/brax/envs/__init__.py", line 63, in get_environment
    return _envs[env_name](**kwargs)
  File "/home/varad/study/robotics/mujoco_sims/mjx_test/server/test_box.py", line 62, in __init__
    sys = mjcf.load_model(model)
  File "/home/varad/venv/mjx/lib/python3.10/site-packages/brax/io/mjcf.py", line 264, in load_model
    raise NotImplementedError(
NotImplementedError: Only joint transmission types are supported for actuators.
```

This clearly shows that even though MJX accepts `RK4` integrator and `site` transmission (if `RK4` is changed to `euler`), `brax` pipeline shows error. I tried some solution of my own, but I thought there might be some subtle things that might break, hence raised this issue. Please let me know if anything more is required. Sorry, i forgot to attach all of this, when raising the issue.

btaba, 2024-02-27:
Hi @varadVaidya 
Looking at the traceback, Line [260](https://github.com/google/brax/blob/main/brax/io/mjcf.py#L260
) and [264](https://github.com/google/brax/blob/3c109cfd131e01691a53891e8f2ec9b32cf97670/brax/io/mjcf.py#L264) in `brax/io/mjcf.py` don't raise any exceptions. Can you try updating your version of brax? See https://github.com/google/brax/releases/tag/v0.10.0

varadVaidya, 2024-02-27:
Ah. good catch, my server had the updated version, while i made the minimal code on a older local version of `brax`. The error still shows up. I have edited the traceback in the previous comment to reflect on the new one.

btaba, 2024-02-27:
The traceback suggests you are using the `generalized` backend. You should set the backend string to "mjx" in `super().__init__` for your `class Box(PipelineEnv):`

See https://github.com/google/brax/blob/3c109cfd131e01691a53891e8f2ec9b32cf97670/brax/envs/base.py#L88

closed 2024-02-27 by varadVaidya