google-deepmind/mujoco #1443: Robot going through objects in environment

santoz007, 2024-02-25:
Hi,

I am trying to simulate hybrid force/position control of a manipulator using mujoco py. In between the simulation, the manipulator penetrates through a fixed object in the environment instead of moving over it i.e. no penetration must take place. This is causing my simulation to stop entirely.   

[image]

My model file is defined as follows:

[image]

Any help is highly appreciated.

kevinzakka, 2024-03-01:
Few recommendations here:

* Increase the thickness of your table.
* Replace the collision geom of the last link of your robot with a primitive like a cylinder or a sphere.
* Reduce your simulation timestep.

r-aristov, 2024-03-04:
> Few recommendations here:
> 
> * Increase the thickness of your table.
> * Replace the collision geom of the last link of your robot with a primitive like a cylinder or a sphere.
> * Reduce your simulation timestep.

Have similar issue with mujoco-mjx here: [Collisions not working properly in mujoco-mjx with Humanoid model](https://github.com/google-deepmind/mujoco/issues/1468)

Unfortunately none of the suggested things worked.

btaba, 2024-03-11:
Hi @r-aristov , the issue referenced for Humanoid was closed, Can you please provide a complete example that we can reproduce? I don't really understand what the expected behavior is vs. the behavior you're seeing, do you have a video?

r-aristov, 2024-03-19:
@btaba Hello!
The issue was resolved, thanks!
It turned out that I missed **conaffinity** parameter and by default collisions were turned on only for feet of Humanoid.

closed 2024-05-10 by btaba