google-deepmind/mujoco #1444: Part of the model is lost in the mujoco

hooyahliu, 2024-02-25:
Hi,

I'm a student and I'm trying to use MuJoCo to add an object mesh.
But part of the object model is lost in the mujoco.

Here is the mjcf I use:

<details>
  <summary>minimal XML</summary>

```XML

<mujoco model='object_pose'>
    <!-- ======= WORLD ======= -->
    <worldbody>
        <!-- ======= HAND ======= -->
        <body name="forearm" pos="0 0 1" euler="0 0 0" >
            ...
        </body>

        <!-- ======= OBJECT_BODY ======= -->
        <body name="Object" pos="0 0 0" euler="0 0 0" >
            <geom name="object" type="mesh" mesh="object_grasp" material="MatWoodB" />
            <joint name="OBJTx" pos="0 0 0" axis="1 0 0" type="slide" limited="false" damping="0" />
        </body>
    </worldbody>
 
    <!-- ======= OBJECT_ASSET ======= -->
    <asset>
        <mesh file="/meshdata/core-can-129880fda38f3f2ba1ab68e159bfb347/coacd/decomposed.obj" name="object_grasp" scale="0.08 0.08 0.08" />
    </asset>
</mujoco>
```

</details>

I tried using methods in [#326 and [#327], but it didn’t solve the problem.

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model='object_pose'>
    <compiler angle='radian' discardvisual="false" /> 
    <!-- ======= WORLD ======= -->
    <worldbody>

...

```

santoz007, 2024-02-26:
Hello,

Try moving the asset tags before the world body tags. This is typically how I structure my model files and it works.

hooyahliu, 2024-02-26:
> Hello,
> 
> Try moving the asset tags before the world body tags. This is typically how I structure my model files and it works.

Thank you for your reply. 
I tried the method you mentioned, but it didn't work. 
But just now I used trimesh to convert the obj file into an stl file, which solved my problem.

closed 2024-02-26 by hooyahliu