google-deepmind/mujoco #1455: unrecognized attribute: 'memory' when running

anupampatil44, 2024-02-26:
**How to ask for help**

I tried running the (model) [particle.xml](https://github.com/google-deepmind/mujoco/blob/main/model/composite/particle.xml) simulation, mentioned in [1158](https://github.com/google-deepmind/mujoco/issues/1158).
Below is the error I got:

```./simulate /home/anupam/mujoco/model/composite/particle.xml
MuJoCo Pro version 2.10
XML Error: Schema violation: unrecognized attribute: 'memory'

Element 'size', line 24, column 3

```

</details>

Please let me know what might be causing the issue.
Thanks!

anupampatil44, 2024-02-27:
Getting this error when I try to run the simulate binary for  version [3.1.2](https://github.com/google-deepmind/mujoco/releases/tag/3.1.2).
It runs perfectly when I switch to [2.3.7](https://github.com/google-deepmind/mujoco/releases/tag/2.3.7).
Kindly look into the same.

kevinzakka, 2024-03-01:
Hi @anupampatil44, the model loads perfectly in 3.1.2. From your error message, it looks like you are using MuJoCo Pro but your follow-up message says otherwise?

anupampatil44, 2024-03-01:
I think it's an issue on my end, works fine now. I think I may have confused the binaries, that's my bad.

closed 2024-03-01 by anupampatil44