google-deepmind/mujoco #1456: [MJX] How to get contact forces with the newest version of MJX (using PipelineEnv)?

AlexS28, 2024-02-27:
Hi,

I am trying to get contact information at the feet of my bipedal robot. While I found this very easy to do in mujoco, which is essentially just looking at data.contact, this 'data' type seems to differ from mujoco vs mjx. In mjx, the data.contact returns none. 

Specifically, as done in the MJX tutorial for the humanoid, in the step function I have this: 

data0 = state.pipeline_state
data = self.pipeline_step(data0, motor_targets)

This data does not have any contact information, it is simply none. Is this issue related to the recent change of going from the MJXEnv to the PipelineEnv? If not, any help would be greatly appreciated in trying to get the contact information. I also tried the code below but it also doesn't work within the step function (I just did it for testing and gives error: result_i.nnzJ = nefc * m.nv
TypeError: (): incompatible function arguments. The following argument types are supported:
    1. (arg0: mujoco._structs.MjData, arg1: int) -> None). So now I am not sure what to do. 
mjx_data = mjx.make_data(self.mj_model)
mj_data = mjx.get_data(self.mj_model, mjx_data)

Sincerely,
Alex

AlexS28, 2024-02-27:
I noticed another person had a similar issue posted on brax regarding the use of PipelineEnv: https://github.com/google/brax/issues/353#issuecomment-1959800068

btaba, 2024-02-28:
This should be fixed in [0825bc](https://github.com/google/brax/commit/0825bcb74b53e36a62d50405a85e540fe2c25a95) , please let us know if that doesn't work! Closing for now

closed 2024-02-28 by btaba

AlexS28, 2024-02-28:
thanks @btaba , I just tested it and I am getting contact now,  even without needing to debug set to true anymore, thanks to your new update change. thank you again

AlexS28, 2024-03-04:
hello @btaba , first sorry for not investigating this issue a bit more closely earlier. But there's a possibility that the pipeline_state.contact, received from the PipelineEnv (after doing something like pipeline_state = pipeline_step(state.pipeline_state, motor_targets) is always constant. I don't see the contact actually changing during the training phase, leading me to believe something is still not correct. Maybe I am doing something wrong, and I am not supposed to be getting contact from pipeline_state.contact? The way I tested it is I created two scenarios, one with the robot starting in the air (which should be no contact), and on the ground (with contact). But pipeline_state is showing contact for both scenarios. Doing it through the mujoco data (without brax/mjx) contact works as expected.

btaba, 2024-03-05:
Hi @AlexS28 , since shapes are static and known at compile-time, the contact will always be calculated, whether or not there is contact. To check if a contact is penetrating, check the `dist` field to see if it is less than zero. Let us know if you think there is still a bug

AlexS28, 2024-03-05:
sounds great, thanks for the clarification. I just tested it now, and contact is now changing so I believe it is working. Checking the dist being less than zero as you mentioned seemed to have been the key.