google-deepmind/mujoco #1458: Can the actuator of muscle support `earlyact`?

Beanpow, 2024-02-28:
The parameter of `actearly` supports using the next value of the activation variable rather than the current one. But I found it only capable of the general actuator, which doesn't support the actuator of muscle. Can the actuator of muscle support `earlyact`?

https://github.com/google-deepmind/mujoco/blob/a5ea87b0e62641de1ac818d03e3378bbe448bdce/src/xml/xml_native_reader.cc#L2053-L2055

nimrod-gileadi, 2024-02-28:
It is possible to make a `muscle` actuator using the `general` tag. If you look at the docs for [`<muscle>`](https://mujoco.readthedocs.io/en/stable/XMLreference.html#actuator-muscle), they list what attributes you need to set on the `<general>` tag to match the behaviour of `<muscle>`.

Does that work for you?

We could add `actearly` to all the actuator shortcuts, but this shouldn't block you for now.

yuvaltassa, 2024-02-28:
NB: If you load a muscle actuator (or any shortcut) and just save the XML, it gets saved as general.

Beanpow, 2024-02-29:
Perfect! Thanks

closed 2024-02-29 by Beanpow