google-deepmind/mujoco #1463: How can I import some action to make robot move and visualize by mujoco GUI?

DDDog-WANG, 2024-03-01:
**How to ask for help**

First, read our quick guide to
[asking good questions](https://github.com/google-deepmind/mujoco#asking-questions).
Below is a template for you to use:

Hi,

I'm a (student / professor / engineer) and I'm trying to use MuJoCo for Robot learning.

I have already trained robot to study actions, and want to visualize by mujoco GUI to confirm actions. From documents, I know how to load robot and environment to mujoco GUI, but I want to know how to import actions and controllers?

kevinzakka, 2024-03-01:
Hi @DDDog-WANG, you can write a controller that writes into `data.ctrl` and sets the signal for the actuators which will drive the joints. Here is some example code for an [inverse kinematics controller](https://github.com/kevinzakka/mjctrl/blob/main/diffik.py), hope it helps.

closed 2024-03-01 by kevinzakka

DDDog-WANG, 2024-03-02:
This seems help me a lot, I will check it. Thx!!