google-deepmind/mujoco #1464: Interaction between flexcomp and composite bodies. Trouble creating flexcomp/cloth.

nhatminh2h, 2024-03-01:
Hi,

I'm a PhD student and I'm trying to use MuJoCo for soft aquatic robot simulation.

I'm looking for some help with simulating a tendon-driven fin. In the context of MuJoCo, the interaction of tendon, composite and flexcomp.

The video is what I am trying to simulate, a tendon-driven actuator embedded in a membrane:

https://github.com/google-deepmind/mujoco/assets/38815213/c2bdc2eb-d285-4d9b-aa48-b8fc4e982c56

I have made a model of the tendon part below, but I cant figure how to make and attach the membrane.

Tendon model, I have only manage to make cable type work. 1D elastic model is accurate enough but is there a more elegant way to control the geom?
<details>

  <summary>minimal XML</summary>

```XML
<mujoco model="flat_belt_tendon">
  <compiler angle="radian"/>
  <option timestep="0.005" integrator="implicit"/>
  <default class="main">
    <joint limited="true" stiffness="100" range="-1 1" damping="1"/>
    <geom size="0.001 0 0" type="capsule" rgba="0.9 0.7 0.1 1"/>
    <site size="0.001 0.005 0.005" rgba="0.9 0.9 0.9 1"/>
    <default class="visual_cylinder">
      <geom size="0.02 0 0" type="cylinder" rgba="0.3 0.9 0.3 0.4"/>
      <site size="0.005 0.005 0.005" rgba="0.5 0.5 0.5 1"/>
    </default>
  </default>
  <extension>
    <plugin plugin="mujoco.elasticity.cable">
      <instance name="composite">
        <config key="twist" value="1e9"/>
        <config key="bend" value="5e7"/>
        <config key="vmax" value="0.6"/>
      </instance>
    </plugin>
    <plugin plugin="mujoco.elasticity.shell"/>
  </extension>
  <custom>
    <text name="composite_" data="rope_"/>
  </custom>
  <worldbody>
    <geom name="root" pos="0 0 0.902" type="sphere" rgba="1 0 0 1"/>
    <geom name="end" pos="0.033 0 0.902" type="sphere" rgba="1 0 0 1"/>
    <site name="root" pos="0 0 0.902"/>
    <light pos="0 0 1" dir="0 0 -1"/>
    <body name="B_first" pos="0 0 0.9">
      <geom name="G0" size="0.000515625 0.003 0.0001" pos="0.000515625 0 0" type="box" group="3" density="900" rgba="0 3.53699e-05 1 1"/>
      <site name="S_first" pos="0 0 0" size="0.005" group="3" rgba="0.5 0.5 0.5 1"/>
      <plugin instance="composite"/>
      <body name="B_1" pos="0.00103125 0 0">
        <joint name="J_1" pos="0 0 0" limited="false" type="ball" group="3" stiffness="0" range="0 0" damping="0.0005"/>
        <geom name="G1" size="0.000515625 0.003 0.0001" pos="0.000515625 0 0" type="box" group="3" density="900" rgba="0 3.42248e-05 1 1"/>
        <plugin instance="composite"/>
        <body name="B_2" pos="0.00103125 0 0">
          <site name="S_2" pos="0 0 0.0017" size="0.005" group="3" rgba="0.5 0.5 0.5 1"/>
          <joint name="J_2" pos="0 0 0" limited="false" type="ball" group="3" stiffness="0" range="0 0" damping="0.0005"/>
          <geom name="G2" size="0.000515625 0.003 0.0001" pos="0.000515625 0 0" type="box" group="3" density="900" rgba="0 3.19733e-05 1 1"/>
          <plugin instance="composite"/>
          <body name="B_3" pos="0.00103125 0 0">
            <joint name="J_3" pos="0 0 0" limited="false" type="ball" group="3" stiffness="0" range="0 0" damping="0.0005"/>
            <geom name="G3" size="0.000515625 0.003 0.0001" pos="0.000515625 0 0" type="box" group="3" density="900" rgba="0 2.97997e-05 1 1"/>
            <plugin instance="composite"/>
            <body name="B_4" pos="0.00103125 0 0">
              <joint name="J_4" pos="0 0 0" limited="false" type="ball" group="3" stiffness="0" range="0 0" damping="0.0005"/>
              <geom name="G4" size="0.000515625 0.003 0.0001" pos="0.000515625 0 0" type="box" group="3" density="900" rgba="0 2.77047e-05 1 1"/>
              <plugin instance="composite"/>
              <body name="B_5" pos="0.00103125 0 0">
[... 298 lines cut ...]
  </default>

  <worldbody>
    <light pos="0 0 5"/>

    <site name="root" pos="0 0 .902"/>
    
    <geom type="sphere" name="root" pos="0 0 .902" rgba="1 0 0 1"/>
    <geom type="sphere" name="end" pos="0.033 0 .902" rgba="1 0 0 1"/>
    
    
    <flexcomp name="membrane" type="grid" dim="2" count="5 4 1" spacing="0.01 0.008 0.002" pos="0.02 0 .9" radius="0.001">
      <plugin plugin="mujoco.elasticity.shell">
        <config key="poisson" value="0"/>
        <config key="thickness" value="1e-5"/>
        <config key="young" value="0.125e6"/>
      </plugin>
    </flexcomp>
    

  </worldbody>
</mujoco>
```

</detail>

yuvaltassa, 2024-03-06:
You are trying some very advanced stuff... we've never tried this ourselves so it's difficult to help you :slightly_smiling_face: 

To your specific question, flexcomps, like composites, will expand to a lower-level representation when saved to XML. In that representation you should be able to attach a tendon. Since I see you did this with the composite I expect that you tried this with the flexcomp as well. Did you try and fail?

nhatminh2h, 2024-03-11:
Thank you for getting back to me. I really enjoy the demos on your youtube channel!

I have tried playing with the generated XML file(attaching sites for the tendon, is there a macro to add these?). I have attached the best attempt below(elasticity.cables + trial and error with the bending stiffness). It's quite promising but I tried the exact thing with grid and it seems to crash mujoco.

https://github.com/google-deepmind/mujoco/assets/38815213/f99c182b-de05-434e-9fc8-40d432493dc7

I will go through all the approaches one more time systematically and get back to you.

wangcongrobot, 2024-03-22:
The video is cool! Any update for the new model?