google-deepmind/mujoco #1466: mjSOL_NEWTON not resulting in speed up in MJX notebook

goncalog, 2024-03-02:
It's possible this isn't a bug but just a misundertanding on my part. Apologies if that's the case.

I've tried replacing the `mjSOL_CG` solver in the Humanoid class in the MJX tutorial notebook with `mjSOL_NEWTON`, but this didn't result in a speed up in training time.
```
class Humanoid(PipelineEnv):

  def __init__(
      self,
      forward_reward_weight=1.25,
      ctrl_cost_weight=0.1,
      healthy_reward=5.0,
      terminate_when_unhealthy=True,
      healthy_z_range=(1.0, 2.0),
      reset_noise_scale=1e-2,
      exclude_current_positions_from_observation=True,
      **kwargs,
  ):
    path = epath.Path(epath.resource_path('mujoco')) / (
        'mjx/test_data/humanoid'
    )
    mj_model = mujoco.MjModel.from_xml_path(
        (path / 'humanoid.xml').as_posix())
    mj_model.opt.solver = mujoco.mjtSolver.mjSOL_NEWTON
```

Release v3.0.1 refers the Newton solver significantly speeds up simulation on GPU:

[image]

Perhaps wrongly but I was expecting the speed up in simulation to improve total training time too - is this expectation misplaced? Or is there something else that I'm missing? 

I've tested this on Colab using a T4 GPU runtime.

Thank you for considering this issue.

yuvaltassa, 2024-03-03:
Did you reduce the [iteration](https://mujoco.readthedocs.io/en/latest/XMLreference.html#option-iterations) count?

Newton iterations are not faster, they converge better (usually 2 or 3 is sufficient). To see the speedup you need to reduce the iteration count.

This is documented [here](https://mujoco.readthedocs.io/en/latest/mjx.html#performance-tuning).

yuvaltassa, 2024-03-03:
Closing under the assumption that this is the issue, but feel free to reply here if I am wrong.

closed 2024-03-03 by yuvaltassa

goncalog, 2024-03-03:
thank you for looking into this @yuvaltassa - much appreciated! i've tried `iterations = 3` as suggested but training still isn't faster (see below). anything else i may try/could be missing?

### original
```
mj_model.opt.solver = mujoco.mjtSolver.mjSOL_CG
mj_model.opt.iterations = 6
mj_model.opt.ls_iterations = 6
```
[image]

### new
```
mj_model.opt.solver = mujoco.mjtSolver.mjSOL_NEWTON
mj_model.opt.iterations = 3
mj_model.opt.ls_iterations = 6
```
[image]