google-deepmind/mujoco #1468: Collisions not working properly in mujoco-mjx with Humanoid model

r-aristov, 2024-03-04:
Hello, I'm an independent researcher working with MuJoCo-MJX to train neural networks for controlling the Humanoid model, the same one utilized in Brax. However, I'm not using Brax. Instead, I've developed my own pipeline, though I don't think that detail is particularly relevant here.

Unfortunately, I'm encountering an issue with collisions; the Humanoid occasionally passes through the floor. Similarly, self-intersections aren't functioning correctly. While the system successfully manages collisions involving the tips of the Humanoid's legs and arms, it struggles with handling collisions at the knees and elbows.
Am I missing something, or this is a bug in simulation?

[Here's the Humanoid model from Brax for reference.](https://github.com/google/brax/blob/a89322496dcb07ac5a7e002c2e1d287c8c64b7dd/brax/envs/assets/humanoid.xml)

[image]

I've attempted to fix the issue by increasing the number of iterations and decreasing the time step tenfold, but unfortunately, these adjustments did not fix the issue.

Feel free to admire the best strategy that MuJoCo Man has mastered so far:

https://github.com/google-deepmind/mujoco/assets/25738725/ec902cb6-b764-4571-8de3-c7c826c312bc

Balint-H, 2024-03-04:
The default humanoid XML has self collisions and non end-effector collisions with the ground disabled. Check the `contype` and `conaffinity` fields in the xml, as well as the relevant parts of the [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-contype). Also check out the `<contact></contact>` section of your xml if it has it, as that can also affect contact behaviour. (Relevant [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact)).

closed 2024-03-04 by yuvaltassa

r-aristov, 2024-03-04:
> The default humanoid XML for MJX has self collisions and non end-effector collisions with the ground disabled. Check the `contype` and `conaffinity` fields in the xml, as well as the relevant parts of the [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-geom-contype). Also check out the `<contact></contact>` section of your xml if it has it, as that can also affect contact behaviour. (Relevant [docs](https://mujoco.readthedocs.io/en/latest/XMLreference.html#contact)).
> 
> By editing these elements (e.g. removing `excludes` from `contact` and setting `contype` and `conaffinity` to 1) you can control what intersects with what.

Amazing, I've completely missed this feature! Thank you so much!