google-deepmind/mujoco #1471: Improving the speed of the mujoco step

girishsmurthyIntel, 2024-03-04:
[assets.zip](https://github.com/google-deepmind/mujoco/files/14484869/assets.zip)
Hi,

I'm a (engineer) and I'm trying to use MuJoCo for multi robot warehouse environment. I'm looking for some help with speed of the mujoco step. We have a warehouse environment with multiple robots and manipulator arms created in Mujoco [older version of 2.1]. When we profile across the **mujoco sim.step(), the python time shows ~80ms/per call**. 
We have checked with testspeed utility as well, which shows 
Running 10 steps at dt = 0.002 ...
 Simulation time      : 0.04 s
 Steps per second     : 253
 Realtime factor      : 0.51 x
 **Time per step        : 3.9603 ms**
 Contacts per step    : 390
 Constraints per step : 1055
 Degrees of freedom   : 242. 
I am not sure which one is correct. We need to further accelerate this, to work in real time. In this regard, we would need to know how this could be done. Would this need multi cores or any specific GPU models?

Here is a model which explains my question:

The xml file is in the attached folder (assets/warehouse_v2.xml)

girishsmurthyIntel, 2024-03-05:
I have updated the model attachments. Request you to kindly check.
[assets_2.zip](https://github.com/google-deepmind/mujoco/files/14489729/assets_2.zip)

yuvaltassa, 2024-03-05:
I see lots of warehouse_Vx.xml's, but none of them load?

girishsmurthyIntel, 2024-03-05:
The xml file to be used is  (assets/warehouse_v2.xml)

yuvaltassa, 2024-03-05:
Using the simulate binary from the 3.1.2 tarball, I get a parsing error related to <include>. Do you not get that?

yuvaltassa, 2024-03-05:
Please make sure it loads in MuJoCo 3.1.2

girishsmurthyIntel, 2024-03-05:
Okay, But we have been using Mujoco 210 and haven't yet migrated to the latest.

yuvaltassa, 2024-03-05:
2.1.0 was released more than 2 years ago and we've made **many** improvements since then, including to performance. Please upgrade to the latest version (at least for this model that you asking us for help with), and we can take it from there.

girishsmurthyIntel, 2024-03-05:
Please find the updated model for mujoco 3.1.2
[Uploading assets_3.zip…]()

yuvaltassa, 2024-03-05:
I think something went wrong with your upload? I am unable to grab this zip file.

girishsmurthyIntel, 2024-03-05:
[assets_3.zip](https://github.com/google-deepmind/mujoco/files/14497611/assets_3.zip)

ykim104, 2024-03-05:
I am looking for a similar advice. I have a large scene with high DoFs. I look forward to the collaborators' replies!

yuvaltassa, 2024-03-05:
Okay so running `testspeed warehouse_v2.xml` (that's the one, yes?), I get

```
Rolling out 10000 steps, at dt = 0.002...

 Simulation time      : 8.08 s
 Steps per second     : 1237
 Realtime factor      : 2.47 x
 Time per step        : 808.2 µs

 Contacts per step    : 147.05
 Constraints per step : 638.70
 Degrees of freedom   : 242

 Internal profiler, µs per step
              step :  808.2  (100.00 %)
           forward :  797.5  ( 98.68 %)
          position :  579.7  ( 71.74 %)
          velocity :   18.0  (  2.22 %)
         actuation :    1.4  (  0.17 %)
        constraint :  194.5  ( 24.07 %)
           advance :    9.8  (  1.21 %)
             other :    4.8  (  0.59 %)

   position total  :  579.7  ( 71.74 %)
     kinematics    :   25.4  (  3.15 %)
     inertia       :   12.5  (  1.54 %)
     collision     :  481.9  ( 59.62 %)
       broadphase  :   85.6  ( 10.59 %)
       narrowphase :  368.3  ( 45.58 %)
     make          :   59.7  (  7.39 %)
     project       :    0.0  (  0.00 %)
```

So realtime factor of 2.5x, and this at dt = 2ms. You can probably increase that to 5ms which would give you a real time factor of 6.2. Now, what's specifically the issue that you want to solve?

ykim104, 2024-03-05:
Is increasing the timestep the only way to improve realtime factor?

girishsmurthyIntel, 2024-03-06:
Q1: Can I keep a profiler across the mujoco.mj_step and measure the latency, will that be similar to the testspeed utility?
       Because earlier, for the same xml file, I had observed that the profiler across the step showed me ~80ms for each step while the test speed numbers were as below
Running 10 steps at dt = 0.002 ...
Simulation time : 0.04 s
Steps per second : 253
Realtime factor : 0.51 x
Time per step : 3.9603 ms
Contacts per step : 390
Constraints per step : 1055
Degrees of freedom : 242.

Q2: How can I accelerate this to run it in real time as my simulation is very slow?

yuvaltassa, 2024-03-06:
@ykim104: No, of course not, it is simply the easiest way. There are many other things one can do, but they are model specific. For example, looking at the `testspeed` printout from @girishsmurthyIntel's model, you can see that nearly 50% of the time is spent in narrow phase collisions. It might be the case that there are lots of tested collisions that never occur. This can be mitigated by using [collision pre-filtering](https://mujoco.readthedocs.io/en/latest/computation/index.html#selection) so that collisions which never occur are not even checked for. Another option is to replace slow collisions (mesh-mesh) with faster collisions (primitive-primitive) by improving the collisions geometry. These are just two examples, there are many other options. I was simply stating the easiest and most obvious. We should probably add a "Performance tuning" section to the documentation...

@girishsmurthyIntel I have no idea what is happening in your code. I've never seen such discrepancy between step time in Python and time in C. Python might add a few microseconds, but not this. I suspect that you are not just timing mj_step, or perhaps you are initializing the state such that all the robots are overlapping, or some other silliness. 

Also, as you can see from our printouts, latest MuJoCo is about 5x faster than the 2.1 version for your model. Perhaps it is time to upgrade? :slightly_smiling_face:

closed 2024-03-06 by yuvaltassa

ykim104, 2024-03-07:
Thank you @yuvaltassa for your answer!! I will check out collision pre-fitlering and see how I can apply it to my case. I am using procedurally generated scenes with custom object meshes so I can't replace mesh-mesh with primitive-primitive. 

I think the section on "Performance tuning" would be very very helpful! Do you have any other suggestions to improve performance (less jittery mesh-mesh and faster compute). I know we can look at the documentation and that it's quite comprehensive, but it would be really helpful to have a list of topics that are more relevant for me to look first.

Thank you so much for your help!

girishsmurthyIntel, 2024-03-07:
Hi @yuvaltassa, Thank you for the analysis. Between I tried to check the mujoco step code invoke and observed that the n_substeps have been configured to 20. This could have caused the scaling of the mujoco step time?
Yes, definitely we will migrate to latest mujoco version of 3.1.0. This indeed requires an effort to move from the mujoco-py to mujoco python bindings.

yuvaltassa, 2024-03-07:
Yes, it would scale it by 20x  🙂 

mujoco-py is old and unsupported, performance is not even the most important reason to migrate away from that.

This might come in handy:

https://mujoco.readthedocs.io/en/latest/python.html#migration-from-mujoco-py

commit b9711096 (2024-03-08): Add Performance Tuning section.

Fixes #1471.

PiperOrigin-RevId: 613867065
Change-Id: I947284300709e0fd6633068d9690a4b19812141f