google-deepmind/mujoco #1472: Discrepancy in mass calculation from mesh

vaxenburg, 2024-03-04:
I'm seeing a significant difference in mass calculation from meshes in two different MuJoCo versions. Here is a minimal example:

```python
from dm_control import mjcf

xml = """
<mujoco>
    <asset>
        <mesh name="thorax" file="thorax_body.msh"/>
    </asset>
    <worldbody>
        <body name="thorax">
           <geom name="thorax" type="mesh" mesh="thorax" density="0.478"/>
        </body>
    </worldbody>
</mujoco>
"""

physics = mjcf.Physics.from_xml_string(xml)
mass = physics.named.model.body_mass['thorax']

print(mass)
# Prints 0.20764519675374468 in mujoco 2.3.1
# Prints 0.22578387429620797 in mujoco 3.1.2
```

This particular `thorax_body.msh` mesh can be found [here](https://github.com/google-deepmind/dm_control/blob/main/dm_control/locomotion/walkers/assets/fruitfly_v2/thorax_body.msh), but the issue is not specific to this mesh and probably any mesh would do.

Any ideas would be very helpful!

quagla, 2024-03-05:
This is caused by the change

https://github.com/google-deepmind/mujoco/commit/ffca553f645b5c2fcfdfffc7eb3f462b271f5368

The old computation inferred the volume of the object from the inertia box whereas we now use the actual volume for computing the mass. Apologies if this was not documented in the changelog.

closed 2024-03-05 by quagla

vaxenburg, 2024-03-14:
Thank you! This is good to know.

commit 49ddb7ca (2024-03-05): Update changelog for 3.1.0.

Fixes #1472.

PiperOrigin-RevId: 612787350
Change-Id: Ia5609266462b5d537ce0c03efa764f1f32bc9389