google-deepmind/mujoco #1473: Inconsistent contacts information between mjx and mujoco when the geoms are a capsule and a box.

Manchewable, 2024-03-05:
Hi, I've found an inconsistency with the data.contact list between mujoco and mjx. I'm wondering if I've stumbled upon a bug or I've did something wrong.

Here's a minimal example that reproduces the problem:
```
import mujoco
from mujoco import mjx

XML=r"""
<mujoco>
<option>
  <flag gravity="disable"/>
</option>

<worldbody>
  <body pos="0 0 0">
    <freejoint/>
    <geom type="box" size=".1 .1 .1" name="box" rgba="1 1 1 0.25"/>
    <site pos="0 0.02500001 0.85" size="0.025" rgba="1 0 0 1"/>
    <site pos="0 0.02500001 1.15" size="0.025" rgba="1 0 0 1"/>
  </body>

  <body pos="0 0 1">
    <freejoint/>
    <geom type="capsule" size=".15 .15" name="capsule" rgba="1 1 1 0.25"/>
  </body>
</worldbody>

<contact>
  <pair geom1="box" geom2="capsule" margin="100"/>
</contact>
</mujoco>
"""

mj_model = mujoco.MjModel.from_xml_string(XML)
model = mjx.device_put(mj_model)
data = mjx.make_data(model)
data = mjx.forward(model, data)

print("mjx:")
print(f"contact.dist: {data.contact.dist}")
print(f"contact.pos: {data.contact.pos}")

mj_data = mujoco.MjData(mj_model)
mujoco.mj_forward(mj_model, mj_data)

print("mujoco:")
print(f"contact.dist: {mj_data.contact.dist}")
print(f"contact.pos: {mj_data.contact.pos}")
```

And here's the output of the above code:
```
mjx:
contact.dist: [1. 1.]
contact.pos: [[0.         0.02500001 0.85      ]
 [0.         0.02500001 1.15      ]]
mujoco:
contact.dist: [0.6 0.9]
contact.pos: [[0.   0.   0.4 ]
 [0.   0.   0.55]]
```

The outputs I get show different contact distances and positions between mjx and mujoco. Here are some screenshots showing the contact positions in red dots, left is mjx and right is mujoco:
[image] [image]

Also, why does it return two contacts with different positions? It seems to me that the actual midpoint is `[0, 0, 0.4]`, why does it return another midpoint with a different position?

Context:
- Operating system: Ubuntu 22.04
- Mujoco version: mujoco-mjx-3.1.2

yuvaltassa, 2024-03-05:
I can answer the second question. Your geoms are so large with the margin, that the box-capsule collider is colliding the capsule as if it is two spheres (imagine a capsule embedded in a box, these collisions are what you want), therefore the two points returned by MuJoCo are the collision points for the two spheres. @btaba to comment about MJX collisions.

Manchewable, 2024-03-06:
Thanks @yuvaltassa! I've just stumbled upon https://github.com/google-deepmind/mujoco/issues/1325 which I think answers my first question.

To follow up on my second question, why is it that if I only move the capsule along the x or y-axis, it'll return only one contact position as opposed to two?
[image]

If I assume that it'll return two different contacts most of the time, and I want to know the contact distance between those two geoms during simulation, would it be right for me to just take the contact with the smaller distance?

yuvaltassa, 2024-03-13:
The images above are for MJX or MuJoCo? I'm not sure I know what you're asking without being able to reproduce.

> If I assume that it'll return two different contacts most of the time, and I want to know the contact distance between those two geoms during simulation, would it be right for me to just take the contact with the smaller distance?

Yes

btaba, 2024-03-14:
Hi @Manchewable , thanks for the bug report. It turns out `margin` wasn't really implemented for capsule-box, and so we just pushed out a change to raise an error if `margin` is present for those collisions in MJX.

closed 2024-08-15 by kevinzakka