google-deepmind/mujoco #1474: Sphere robot trajectory is not expected.

LinShuang22, 2024-03-05:
Hi, @yuvaltassa 
I see the earlier issue #1310. I have a modeling issue similar to his, and I discussed it with you in that previous question.

My model is a spherical robot driven by an internal differential drive system. To avoid internal collisions, following the suggestions from the previous issue, I've made the car as a direct child body of the spherical shell. Now, the overall motion is not driven by the actual wheels and internal friction of the spherical shell. Instead, I directly make the internal car rotate around the sphere's center to achieve roll and turn, consequently driving the motion of the spherical shell.

I attempted to link the sum and difference of the two wheels with the overall turn and roll joints of the internal car using `equality` constraints. This visually gives the impression that the wheels are rolling internally.

This looks very promising, and everything behaves normally when the turn and roll joints run individually. However, when I move both joints simultaneously, the trajectory of the sphere deviates from expectations. For instance, when both turn and roll are in motion, such as at turn=0.0001 and roll=0.002 (values adjusted via `gear`), the trajectory of the sphere exhibits a wavelike pattern instead of the expected circular motion.

This appears to be a sequencing issue with the turn and roll joints. The joints of the car are not bound to its own coordinate system; when the roll joint rotates, the turn joint doesn't change accordingly. 

How can I adjust the model to make the trajectory of the sphere match the expectations?

<details>
  <summary>minimal XML</summary>

```XML
<mujoco model="FreebotSimplified V1">

  <visual>
    <scale camera="0.1" contactwidth="0.1"
           framelength="0.2" framewidth="0.02"
           jointlength="0.2" jointwidth="0.02"
    />
    <rgba camera="1 0 0 0.6" />
  </visual>

  <size njmax="500" nconmax="100"/>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="2048"/>
    <global offwidth="800" offheight="800"/>
  </visual>

  <default>
    <joint damping=".01"/>
<!--    <geom solref="0.01 1" solimp=".95 .99 .0001" condim="1" />-->
    <default class="decor">
      <site type="box" rgba=".5 1 .5 1"/>
    </default>
    <default class="internal">
      <geom contype="0" conaffinity="0"/>
    </default>
  </default>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="ground" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>
    <camera name="fixed" pos="0 0 1" zaxis="0 -1 0"/>

    <light pos="0 0 6"/>
    <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>

    <body name="shell" pos="0 0 0.041">
      <joint name="shell_joint" pos="0 0 0" type="free"/>
      <geom name="shell_geom" type="sphere" size="0.041" mass="5.0e-02" rgba="0.2 0.8 0.8 0.5"/>
      <site name="shell_site1" rgba="1 0 0 1" pos="0 0 0.041" type="sphere" size="0.003"/>
      <site name="shell_site2" rgba="1 0 0 1" pos="-0.041 0 0" type="sphere" size="0.003"/>

      <body name="tro_base" pos="0 0 -0.011" childclass="internal">
<!--        <site name="roll_center" rgba="1 0 0 1"/>-->
        <joint name="turn" axis="0 0 1" pos="0 0 0.011" />
        <joint name="roll" axis="1 0 0" pos="0 0 0.011" />

        <geom name="chassis_geom" pos="0 0 0.007" mass="1.0e-02" type="box" rgba="0.8 0.9 0.8 1" size="0.020 0.020 0.003" />

        <geom name="chassis_geom2" pos="-0.032 0 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.002 0.02 0.003" />
        <geom name="chassis_geom3" pos="0.032 0 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.002 0.02 0.003" />
        <geom name="chassis_geom4" pos="-0.025 -0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom5" pos="-0.025 0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom6" pos="0.025 -0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom7" pos="0.025 0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />

        <geom name="chassis_geom8" pos="0.015 0 0.020" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.003 0.003 0.010" />
        <geom name="chassis_geom9" pos="-0.015 0 0.020" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.003 0.003 0.010" />
        <site name="chassis_site" rgba="1 0 0 1" pos="0 0.02 0.007" type="sphere" size="0.003"/>

        <body name="chassis1" pos="0 0 -0.013">
          <geom name="chassis1_geom" mass="8.0e-02" type="cylinder" rgba="0.66667 0 1 1" size="0.015 0.013" />
          <geom name="chassis1_geom2" pos="0 0 0.015" mass="1.0e-09" type="box" rgba="0.66667 0 1 1" size="0.017 0.017 0.002" />
        </body>

        <body name="wheel_L" pos="-0.025 0 0" zaxis="1 0 0">
          <joint name="joint_L" pos="0 0 0" axis="0 0 -1" />
          <geom name="left_wheel_geom" mass="5.0e-03" type="cylinder" rgba="1 0.24314 0.023529 1" size="0.0165 0.0035" />
            <site class="decor" size="0.014 0.003 0.0035"/>
            <site class="decor" size="0.003 0.014 0.0035"/>
        </body>

        <body name="wheel_R" pos="0.025 0 0" zaxis="1 0 0">
          <joint name="joint_R" pos="0 0 0" axis="0 0 -1" />
          <geom name="right_wheel_geom" mass="5.0e-03" type="cylinder" rgba="1 0.24314 0.023529 1" size="0.0165 0.0035" />
            <site class="decor" size="0.014 0.003 0.0035"/>
            <site class="decor" size="0.003 0.014 0.0035"/>
        </body>
      </body>

    </body>

  </worldbody>

  <tendon>
    <fixed name="turn">
      <joint joint="turn" coef="1"/>
    </fixed>

    <fixed name="roll">
      <joint joint="roll" coef="1"/>
    </fixed>

    <fixed name="sum">
      <joint joint="joint_L" coef=".5"/>
      <joint joint="joint_R" coef=".5"/>
    </fixed>

    <fixed name="difference">
      <joint joint="joint_L" coef="-.5"/>
      <joint joint="joint_R" coef=".5"/>
    </fixed>
  </tendon>

  <equality>
    <tendon tendon1="sum" tendon2="roll" polycoef="0 1.6753 0 0 0"/>
    <tendon tendon1="difference" tendon2="turn" polycoef="0 1 0 0 0"/>
  </equality>

  <contact>
    <pair name="shell1-ground" geom1="ground" geom2="shell_geom" friction="1 1 0.9 0.0001 0.0001" />
  </contact>

  <actuator>
    <motor name="turn" tendon="turn" ctrlrange="-0.0001 0.0001" gear="85.2386"/>
    <motor name="roll" tendon="roll" ctrlrange="-0.002 0.002" gear="26.4457"/>
  </actuator>

</mujoco>

```

</details>

Here is a screenshot / video, illustrating my question:
turn=0.0001, roll=0.002
[image]

https://github.com/google-deepmind/mujoco/assets/119031680/9f8f80dd-d42f-4301-b9f2-8d5bf0755447

Balint-H, 2024-03-14:
This bot seems to work well with a single ball joint instead. Setting up actuators for ball joints is sometimes tricky, but in this case they seem to be doing what you are looking for. As far as the wheel rolling illusion goes, I hope it can be configured again with some tendon approach (or via control code). Let me know if I misunderstood the desired behaviour.

<details>
<summary> Modified XML </summary>

```xml
<mujoco model="FreebotSimplified V1">
  <visual>
    <scale camera="0.1" contactwidth="0.1"
           framelength="0.2" framewidth="0.02"
           jointlength="0.2" jointwidth="0.02"
    />
    <rgba camera="1 0 0 0.6" />
  </visual>

  <size njmax="500" nconmax="100"/>

  <visual>
    <headlight diffuse="0.6 0.6 0.6"  ambient="0.3 0.3 0.3" specular="0 0 0"/>
    <rgba haze="0.15 0.25 0.35 1"/>
  </visual>

  <asset>
    <texture type="skybox" builtin="gradient" rgb1="0.3 0.5 0.7" rgb2="0 0 0" width="512" height="3072"/>
    <texture type="2d" name="groundplane" builtin="checker" mark="edge" rgb1="0.2 0.3 0.4" rgb2="0.1 0.2 0.3" markrgb="0.8 0.8 0.8" width="300" height="300"/>
    <material name="groundplane" texture="groundplane" texuniform="true" texrepeat="5 5" reflectance="0.2"/>
  </asset>

  <visual>
    <map force="0.1" zfar="30"/>
    <rgba haze="0.15 0.25 0.35 1"/>
    <quality shadowsize="2048"/>
    <global offwidth="800" offheight="800"/>
  </visual>

  <default>
    <joint damping=".01"/>
<!--    <geom solref="0.01 1" solimp=".95 .99 .0001" condim="1" />-->
    <default class="decor">
      <site type="box" rgba=".5 1 .5 1"/>
    </default>
    <default class="internal">
      <geom contype="0" conaffinity="0"/>
    </default>
  </default>

  <worldbody>
    <light pos="0 0 1.5" dir="0 0 -1" directional="true"/>
    <geom name="ground" size="0 0 0.25" type="plane" material="groundplane" condim="3"/>
    <camera name="fixed" pos="0 0 1" zaxis="0 -1 0"/>

    <light pos="0 0 6"/>
    <light directional="false" diffuse=".2 .2 .2" specular="0 0 0" pos="0 0 5" dir="0 0 -1" castshadow="false"/>

    <body name="shell" pos="0 0 0.041">
      <joint name="shell_joint" pos="0 0 0" type="free"/>
      <geom name="shell_geom" type="sphere" size="0.041" mass="5.0e-02" rgba="0.2 0.8 0.8 0.5"/>
      <site name="shell_site1" rgba="1 0 0 1" pos="0 0 0.041" type="sphere" size="0.003"/>
      <site name="shell_site2" rgba="1 0 0 1" pos="-0.041 0 0" type="sphere" size="0.003"/>
	  <body name="outer_frame">
	  
	  <inertial pos="0 0 0" mass="0.001" diaginertia="0.0000001 0.0000001 0.0000001"/>
      <body name="tro_base" pos="0 0 -0.011" childclass="internal">
<!--        <site name="roll_center" rgba="1 0 0 1"/>-->
		<joint name="ball" type="ball" pos="0 0 0.011" armature="0.00001" />

        <geom name="chassis_geom" pos="0 0 0.007" mass="1.0e-02" type="box" rgba="0.8 0.9 0.8 1" size="0.020 0.020 0.003" />

        <geom name="chassis_geom2" pos="-0.032 0 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.002 0.02 0.003" />
        <geom name="chassis_geom3" pos="0.032 0 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.002 0.02 0.003" />
        <geom name="chassis_geom4" pos="-0.025 -0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom5" pos="-0.025 0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom6" pos="0.025 -0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />
        <geom name="chassis_geom7" pos="0.025 0.0185 0.007" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.005 0.0015 0.003" />

        <geom name="chassis_geom8" pos="0.015 0 0.020" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.003 0.003 0.010" />
        <geom name="chassis_geom9" pos="-0.015 0 0.020" mass="1.0e-09" type="box" rgba="0.8 0.9 0.8 1" size="0.003 0.003 0.010" />
        <site name="chassis_site" rgba="1 0 0 1" pos="0 0.02 0.007" type="sphere" size="0.003"/>

        <body name="chassis1" pos="0 0 -0.013">
          <geom name="chassis1_geom" mass="8.0e-02" type="cylinder" rgba="0.66667 0 1 1" size="0.015 0.013" />
          <geom name="chassis1_geom2" pos="0 0 0.015" mass="1.0e-09" type="box" rgba="0.66667 0 1 1" size="0.017 0.017 0.002" />
        </body>

        <body name="wheel_L" pos="-0.025 0 0" zaxis="1 0 0">
          <joint name="joint_L" pos="0 0 0" axis="0 0 -1" />
          <geom name="left_wheel_geom" mass="5.0e-03" type="cylinder" rgba="1 0.24314 0.023529 1" size="0.0165 0.0035" />
            <site class="decor" size="0.014 0.003 0.0035"/>
            <site class="decor" size="0.003 0.014 0.0035"/>
        </body>

        <body name="wheel_R" pos="0.025 0 0" zaxis="1 0 0">
          <joint name="joint_R" pos="0 0 0" axis="0 0 -1" />
          <geom name="right_wheel_geom" mass="5.0e-03" type="cylinder" rgba="1 0.24314 0.023529 1" size="0.0165 0.0035" />
            <site class="decor" size="0.014 0.003 0.0035"/>
            <site class="decor" size="0.003 0.014 0.0035"/>
        </body>
      </body>
		</body>
    </body>

  </worldbody>

  <contact>
    <pair name="shell1-ground" geom1="ground" geom2="shell_geom" friction="1 1 0.9 0.0001 0.0001" />
  </contact>

  <actuator>
    <motor name="pitch" joint="ball" ctrlrange="-0.0005 0.0005" gear="85.2386 0 0"/>
    <motor name="roll" joint="ball" ctrlrange="0 0" group="4" gear="0 26.4457 0"/>
    <motor name="yaw" joint="ball" ctrlrange="-0.002 0.002" gear="0 0 26.4457"/>
  </actuator>

</mujoco> 
```
</details>

https://github.com/google-deepmind/mujoco/assets/41113387/a12090f4-c1f4-411e-a8e1-510495162b3a

LinShuang22, 2024-03-16:
@Balint-H Thank you! The trajectory of the model you've demonstrated is indeed what I'm looking for! I learned from the XML Reference that the method of connecting the motion of the internal chassis with two wheels using `tendon` and `equality` seems to only apply to hinge and slide joints, and is not suitable for ball joints. Does this mean I cannot achieve the linkage between the wheels and the chassis in the XML file? Do I have to implement wheel animations through a Python script?

Balint-H, 2024-03-22:
An Important thing I wanted to also mention about the ball joint approach, is that it also permits rotations in the direction normal to the wheel surfaces, which would not normally happen a lot. This can partially be mitigated by adding a velocity acutator like this:

`<velocity name="roll_resist" joint="ball" ctrlrange="0 0" group="4" gear="0 0.1 0" kv="0.1"/>` (might need to play with params)
A velocity actuator is used as it will always be in the child body's reference frame, which is why the ball joint actuation works as well. I'll ask around to see if its possible to constrain it in another way.

closed 2024-08-30 by kevinzakka