google-deepmind/mujoco #1476: AMD, Ubuntu, GLFW and mouse, that does not work

iiborisov, 2024-03-05:
I'm looking for some help with visualization using GLFW library. 

My models are simulated and rendered correctly, but the viewer window does not respond to mouse and keyboard inputs.  I have attempted the default mujoco viewer, mujoco python viewer, and a custom GLFW-based viewer, but the issue persists. I suspect the issue lies with my AMD graphics card.

I am running Ubuntu 22.04.4 on an AMD Radeon RX 6600M graphics card and AMD Ryzen 7 5800H processor. When running Ubuntu normally, the mouse and keyboard do not function, and for some reason, the title box becomes black with the title written in bold.
[mouse_does_not_work.webm](https://github.com/google-deepmind/mujoco/assets/62997120/a9a7c867-b7c4-4fb8-8e3d-295cd92b705e)

When I run Ubuntu in recovery mode, however, everything returns to normal, and I am able to interact with the model using a mouse and keyboard. Additionally, the title bar becomes white with a default font.
[mouse_does_work.webm](https://github.com/google-deepmind/mujoco/assets/62997120/b56b2992-c504-45b9-ba71-b0c6a0730bb8)

After recovery mode “some graphic drivers require a full graphical boot and so will fail when resuming from recovery”. It appears that something has been disabled after recovery mode, allowing the use of the viewer with mouse interaction.

This issue did not occur a few weeks ago.

Could you please assist me in resolving this issue? I would appreciate any suggestions or guidance you may have. Thank you for your attention!

saran-t, 2024-03-18:
There's a long-standing issue with the GUI when running in Windows with AMD hardware, but I believe this is the first time we have a complaint from a Linux user...

Before we dig too deep, could you please check if the issue first appears at commit eb844a4b7725822c609ef6a3403c0f1c5635e693 for you?

yuvaltassa, 2024-07-16:
Duplicate of #639

closed 2024-08-30 by kevinzakka