google-deepmind/mujoco #1482: mjx.put_model() takes forever to load a large scene

ykim104, 2024-03-06:
Hi, I'm an engineer and I'm trying to use MuJoCo for learning navigation and manipulation tasks in a scene with many many custom objects.

I am trying out mjx and looking for some help with loading model using `mjx.put_model()` function.
It seems that some functions to check for NotImplementedError and getting mesh attributes from the model takes forever 
( collision_driver.collision_candidates(), mesh.get(), etc)

Is there a way to speed up the loading process?

Thank you for your help!

btaba, 2024-03-11:
Hi @ykim104 , `mesh.get` processes meshes for collision in MJX. If you'd like to avoid all meshes from being processed, please update the `contype` `conaffinity` of the geometries with those meshes. See here:

https://github.com/google-deepmind/mujoco/blob/8413a615d75bf0a0c7edf558f8c089c890732d38/mjx/mujoco/mjx/_src/mesh.py#L316

btaba, 2024-05-23:
Closing the issue do to inactivity, please open a new issue if you are facing any issues!

closed 2024-05-23 by btaba