google-deepmind/mujoco #1485: Improving the speed of the render

girishsmurthyIntel, 2024-03-07:
Hello Team,

        We have a warehouse environment in Mujoco with external cameras.
<body name="side_camera0" pos="-1.5 -0.8 0.3" quat="0.5 0.5 -0.5 -0.5">
<camera euler="0 0 0" fovy="45" name="side_camera0" pos="0 0 0"></camera>

I would need to render and read the depth & RGB data pixels from the camera for some algorithms. Using a profiler (python time.time()) across the render function "self.camera[cam_id].render()", shows ~60ms per call. Can you kindly let me know/suggest ways the rendering from these external cameras in the environment could be accelerated?

yuvaltassa, 2024-03-07:
You need to use a GPU. This timing implies that your backend is a software renderer. You can look at any of our colabs to see how they require a GPU and install the EGL backend.

closed 2024-03-07 by yuvaltassa

girishsmurthyIntel, 2024-03-07:
Thank you, We have used a GPU. I am not sure about the EGL backend. I will check and get back.

yuvaltassa, 2024-03-07:
GPU rendering is consistently < 1ms / call.

Andrew-Luo1, 2024-03-10:
I'd like to add that although MJX doesn't support RGB rendering, you can use the MJX implementation of mj_ray for fast depth rendering. For example, see [here](https://github.com/Andrew-Luo1/jax_shac/blob/main/vision/2dof_ball.ipynb). I've uploaded side-by-side ground-truth mujoco renderings and depth-camera renderings.

Thanks to vmap-batching, MJX scales well if you're doing depth-rendering on parallel RL environments, and probably also if you're rendering high-resolution images. In the above notebook, I'm getting over 3 million FPS depth rendering (8000 parallel environments; 100x100 resolution; NVIDIA 3060 Ti GPU; 4 geoms).

[image]
[image]

girishsmurthyIntel, 2024-03-13:
Dear @yuvaltassa 
         I checked that the nvidia-smi tool shows the mujoco simulation process (running inside of a docker) running on the GPU. ( the usage varies between 30-40%) The docker has direct access to the GPU. Do you have any suggestions to know if the GPU is really utilized for the rendering or there is a CPU bound ?

yuvaltassa, 2024-03-14:
If you getting 60ms per render call there is no need to do any further investigation. You are definitely NOT rendering on the GPU. Take a look at our [public Python notebooks](https://github.com/google-deepmind/mujoco?tab=readme-ov-file#getting-started) they all have a chunk at the top Imports cell that validates that GPU is indeed used for rendering. Perhaps this can help you.

girishsmurthyIntel, 2024-03-25:
Hi @yuvaltassa ,
     I had setup the external cameras as self.camera = {i: **mujoco_py.MjViewer**(self.conveyor.sim) for i in self.camera_ids}
                camera_id.cam.type = const.CAMERA_FIXED
                camera_id._hide_overlay = True
                camera_id._show_mocap = False
                camera_id.cam.fixedcamid = idx 
                glfw.restore_window(camera_id.window)
                glfw.set_window_size(camera_id.window, self.image_width, self.image_height)

A render of this camera was taking ~40-60ms, with a GPU enabled (GPU usage: 30-40%)

Now, I set up the cameras as: self.camera = {i: mujoco_py.**MjRenderContextWindow**(self.conveyor.sim) for i in self.camera_ids} with the same properties as above.

A render of this camera now takes ~1- 1.5ms (GPU usage : 9- 15%)

Any observations on this?