google-deepmind/mujoco #1486: Obtain manipulability measure

aleada23, 2024-03-07:
Hi,

I'm a student and I'm trying to use MuJoCo for control a Baxter robot with Robosuite .

I'm looking for some help with manipulability measure. I find the Jacobian matrix of the robot in function of the joint's variables using sympy. Then I computed the manipulability measure and derived it in order to find the velocities to apply to the joints. However, when I try to evaluate the obtained function in a certain configuration, it takes so much time for the first computation and then the simulation crashes during a second computation. The robot in particular has 7 dof, so the resulting Jacobian has a lot of terms.
Is there a way to better perform calculations or to obtain the manipulability measure for a certain configuration, in order to speed up the simulation?

Here is a model which explains my question:
---first I compute the Jacobian as J and w is the manipulability measure.--
        arg=J*J_trans
	determinant=arg.det(method='lu') 
	w=sp.sqrt(determinant)
        qo_dot=sp.zeros(7,1)
	
	qo_dot[0]=sp.Matrix([[w.diff(q1)]])
	
	qo_dot[1]=sp.Matrix([[w.diff(q2)]])
	
	qo_dot[2]=sp.Matrix([[w.diff(q3)]])

	qo_dot[3]=sp.Matrix([[w.diff(q4)]])

	qo_dot[4]=sp.Matrix([[w.diff(q5)]])

	qo_dot[5]=sp.Matrix([[w.diff(q6)]])

	qo_dot[6]=sp.Matrix([[w.diff(q7)]])

--then at each iteration i perform the following calculations, being actual_q the actual values of the joints--
    q1, q2, q3, q4, q5, q6, q7 = sp.symbols('q1 q2 q3 q4 q5 q6 q7')
    q_dot_num=Ko*np.array(q_dot.subs({q1: actual_q[0],q2: actual_q[1],q3: actual_q[2],q4: actual_q[3],q5: actual_q[4],q6: actual_q[5],q7: actual_q[6],}).evalf()).astype(float)

Thanks, 
Alessandro

yuvaltassa, 2024-03-07:
Why are you using sympy when MuJoCo computes super fast analytic Jacobians (literally in microseconds)?
https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-jac

kevinzakka, 2024-03-07:
@aleada23 See https://github.com/google-deepmind/mujoco/issues/411.

aleada23, 2024-03-08:
I checked #411 but I'm not able to find a solution, sorry. 
I'll try to better explain my problem.
I computed the Jacobian and the function called manipulability measure: 
[image]
After that in order to find the desired joionts' velocities I did:
[image].
It takes long time to do computation so I saved the resulting q_zero_dot using pickle so I don't need to do calculations each time I start the simulation.
Everything was done considering the symbolic values, so at each step of my robot I can obtain the resulting velocities simply substituting the joint variables with their actual value. Here I have the problem, the substitution of symbolic values with reals takes too long at each step.
How can I modify this process to save time during simulation? Is it possible to quickly derive the velocities at each step?
Or can I evaluate the Jacobian with actual value first and then do the derivative of w without symbolic values?
Thank you

yuvaltassa, 2024-03-08:
You don't need any of the symbolic computation. Stop doing that.

closed 2024-03-08 by yuvaltassa

aleada23, 2024-03-08:
I was wondering how to do this without symbolic... if anyone can explain or tell me if there are any examples