google-deepmind/mujoco #1489: Failed to build on Linux

martantoine, 2024-03-08:
I am encountering an error when trying to build from source mujoco, following [this page](https://mujoco.readthedocs.io/en/latest/programming/#building-from-source)  instruction

The command `cmake $PATH_TO_MOJOCO` runs fine
The command `cmake --build .` stops with the following error:
```
[  8%] Building C object CMakeFiles/mujoco.dir/src/engine/engine_io.c.o
In function ‘bufread’,
    inlined from ‘mj_loadModel’ at ........./mujoco/src/engine/engine_io.c:786:3:
........./mujoco/src/engine/engine_io.c:380:3: error: ‘memcpy’ offset [0, 299] is out of the bounds [0, 0] [-Werror=array-bounds=]
  380 |   memcpy(dest, (char*)buf + *ptrbuf, num);
      |   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: all warnings being treated as errors
gmake[2]: *** [CMakeFiles/mujoco.dir/build.make:272: CMakeFiles/mujoco.dir/src/engine/engine_io.c.o] Error 1
gmake[1]: *** [CMakeFiles/Makefile2:1595: CMakeFiles/mujoco.dir/all] Error 2
gmake: *** [Makefile:146: all] Error 2
```

My system is:
- Fedora 39
- gcc (GCC) 13.2.1 20231205 (Red Hat 13.2.1-6)
- cmake version 3.27.7

I am trying the build on the main branch, commit ID: **2ba80206ab5a6468504986b143d83b15dfaa130f**

Thank you for your help

Qrpucp, 2024-03-09:
same issue
gcc 12.3.1 & cmake 3.28.3

Qrpucp, 2024-03-17:
Remove `-Werror` compile option in `cmake/MujocoOptions.cmake` can solve this issue temporarily.
I found this solution in [AUR](https://aur.archlinux.org/packages/mujoco).

martantoine, 2024-03-18:
Thanks for the patch @Qrpucp 
Devs should fix this properly though

saran-t, 2024-03-18:
I've done a quick research on this and it looks like there are several occurrences (outside of MuJoCo) where GCC 13 throws up false positives for array bounds checking.

Given that we do not use GCC internally, this won't be a priority for us to look into. I'm afraid our official stance for the time being will be that GCC 13 is officially "unsupported".

Having said that, we welcome a PR to silence this specific error under GCC 13.