google-deepmind/mujoco #1492: Unable to Build an XML file that reads in a binary MSH file?

kurtenkera, 2024-03-09:
Hi,

I'm a student and I'm trying to use MuJoCo for simulating robots. As I'm new to MuJoCo, I'm just attempting to create an XML file from a basic binary MSH file. However, I cannot achieve even this simple task and I'm unsure why?

Here is the code that I'm currently trying to run in a Jupyter Notebook cell:

```
import mujoco
import mediapy 

xml = r"""
<mujoco>
  <asset>
    <mesh file="hollow_cylinder_binary.msh"/>
  </asset>
  <worldbody>
    <body>
      <geom type="mesh" name="hollow_cylinder_binary" mesh="hollow_cylinder_binary"/>
    </body>
  </worldbody>
</mujoco>
"""

# Make model and data
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

# Make renderer, render and show the pixels
renderer = mujoco.Renderer(model)
mujoco.mj_forward(model, data)
renderer.update_scene(data)

mediapy.show_image(renderer.render())
```

This is the error message I get when I run it:

```
ValueError: Error: invalid sizes in MSH file 'hollow_cylinder_binary.msh'
Object name = hollow_cylinder_binary, id = 0, line = 4
```

I've also attached an image of the `hollow_cylinder_binary.msh' file when visualized in Gmsh:

[image]

Thanks in advance for any help or feedback!

mohammad200h, 2024-03-11:
Hi,
Could you please send us the hollow_cylinder_binary.msh so we can replicate the issue?

mohammad200h, 2024-03-11:
I was able to replicate the issue. Can I ask what you are trying to do with this model? are you trying to simulate a softbody?

TAFFI98, 2024-03-15:
> Hi,
> 
> I'm a student and I'm trying to use MuJoCo for simulating robots. As I'm new to MuJoCo, I'm just attempting to create an XML file from a basic binary MSH file of a hollow cylinder with the available Python bindings. However, I cannot achieve even this simple task and I'm unsure why?
> 
> Here is the code that I'm currently trying to run in a Jupyter Notebook cell:
> 
> ```
> import mujoco
> import mediapy 
> 
> xml = r"""
> <mujoco>
>   <asset>
>     <mesh file="hollow_cylinder_binary.msh"/>
>   </asset>
>   <worldbody>
>     <body>
>       <geom type="mesh" name="hollow_cylinder_binary" mesh="hollow_cylinder_binary"/>
>     </body>
>   </worldbody>
> </mujoco>
> """
> 
> # Make model and data
> model = mujoco.MjModel.from_xml_string(xml)
> data = mujoco.MjData(model)
> 
> # Make renderer, render and show the pixels
> renderer = mujoco.Renderer(model)
> mujoco.mj_forward(model, data)
> renderer.update_scene(data)
> 
> mediapy.show_image(renderer.render())
> ```
> 
> This is the error message I get when I run it:
> 
> ```
> ValueError: Error: invalid sizes in MSH file 'hollow_cylinder_binary.msh'
> Object name = hollow_cylinder_binary, id = 0, line = 4
> ```
> 
> It seems like the error is related to my .msh file. I've attached an image of the `hollow_cylinder_binary.msh' file when visualized in Gmsh below (it is the shape that I want so I'm unsure why MuJoCo cannot read in this file?):
> 
> ![image](https://private-user-images.githubusercontent.com/127472392/311424526-3de7481a-48aa-4f46-9efd-00bead852235.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MTA1MTUxNzgsIm5iZiI6MTcxMDUxNDg3OCwicGF0aCI6Ii8xMjc0NzIzOTIvMzExNDI0NTI2LTNkZTc0ODFhLTQ4YWEtNGY0Ni05ZWZkLTAwYmVhZDg1MjIzNS5wbmc_WC1BbXotQWxnb3JpdGhtPUFXUzQtSE1BQy1TSEEyNTYmWC1BbXotQ3JlZGVudGlhbD1BS0lBVkNPRFlMU0E1M1BRSzRaQSUyRjIwMjQwMzE1JTJGdXMtZWFzdC0xJTJGczMlMkZhd3M0X3JlcXVlc3QmWC1BbXotRGF0ZT0yMDI0MDMxNVQxNTAxMThaJlgtQW16LUV4cGlyZXM9MzAwJlgtQW16LVNpZ25hdHVyZT0zZDZlMmMwMWUzZTI3ZGE4ZjdlZjQ0NGQxYmM4MzhiNDIxNWYyMzMzNTgzZmQwNDA2NzAxMjAzZGQzNDM5NTYxJlgtQW16LVNpZ25lZEhlYWRlcnM9aG9zdCZhY3Rvcl9pZD0wJmtleV9pZD0wJnJlcG9faWQ9MCJ9.hgzxDF63Vu-07sJAXnBA2ox8ncpiHFYgy89GR7DKVzU)
> 
> Thanks in advance for any help or feedback!

Same problem here!

mohammad200h, 2024-03-19:
@TAFFI98 gmsh format is used for flexcomp in Mujoco: https://mujoco.readthedocs.io/en/latest/XMLreference.html#body-flexcomp
Your file need to have a single entity and the xml should look something like this: 

`
<mujoco model="shark">
  <option timestep="0.01" integrator="implicitfast" solver="CG"/>

  <worldbody>
    <flexcomp name="shark" type="gmsh" dim="3" rgba=".8 .2 .2 1" radius="0.001" pos="0 0 .3" mass="10" scale="1 1 1" file="shark.msh">
    <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
`

mohammad200h, 2024-03-22:
closing the issue due to inactivity. Feel free to open it :)

kurtenkera, 2024-03-24:
Hi @mohammad200h, I tried following your advice above. Below is the code cell block that I tried running in a Jupyter notebook environment:

```
import mujoco
import mediapy as media
import matplotlib.pyplot as plt

import time
import itertools
import numpy as np

xml = r"""
<mujoco model="hollow_cylinder_binary">
  <worldbody>
    <flexcomp name="hollow_cylinder_binary" type="gmsh" dim="3" file="mesh_files/hollow_cylinder_binary.msh">
    <edge equality="true"/>
    </flexcomp>
  </worldbody>
</mujoco>
"""

# Make model and data
model = mujoco.MjModel.from_xml_string(xml)
data = mujoco.MjData(model)

# Make renderer, render and show the pixels
renderer = mujoco.Renderer(model)
mujoco.mj_forward(model, data)
renderer.update_scene(data)

media.show_image(renderer.render())
```

As you can see above, I modified the XML file following your advice, however, I got the following error message when I ran the code above:

```
---------------------------------------------------------------------------
ValueError                                Traceback (most recent call last)
Cell In[6], [line 20](vscode-notebook-cell:?execution_count=6&line=20)
      [9](vscode-notebook-cell:?execution_count=6&line=9) xml = r"""
     [10](vscode-notebook-cell:?execution_count=6&line=10) <mujoco model="hollow_cylinder_binary">
     [11](vscode-notebook-cell:?execution_count=6&line=11)   <worldbody>
   (...)
     [16](vscode-notebook-cell:?execution_count=6&line=16) </mujoco>
     [17](vscode-notebook-cell:?execution_count=6&line=17) """
     [19](vscode-notebook-cell:?execution_count=6&line=19) # Make model and data
---> [20](vscode-notebook-cell:?execution_count=6&line=20) model = mujoco.MjModel.from_xml_string(xml)
     [21](vscode-notebook-cell:?execution_count=6&line=21) data = mujoco.MjData(model)
     [23](vscode-notebook-cell:?execution_count=6&line=23) # Make renderer, render and show the pixels

ValueError: XML Error: Error: All nodes must be in single block
Element 'flexcomp', line 4
```

Here is the .msh file that I'm trying to use: 
[hollow_cylinder_binary.zip](https://github.com/google-deepmind/mujoco/files/14737057/hollow_cylinder_binary.zip)

I desperately need to import this binary .msh file into MuJoCo for this project I'm doing. I've been able to import a binary STL file, however, the binary msh file approach is not working! 

Thanks in advance!

mohammad200h, 2024-03-25:
Hi, 
Mujoco supports a single  entity. Your model has 18 points 24 curves 10 surfaces and 1 volume. In order for Mujoco to read this you need to remove points, curves and surface. Only leaving the volume. You can these information by:
1. loading your model in GMSH app 
2. Click on Tools -> Statistics
3. In the Statistics window click on Geometry.

closed 2024-08-30 by kevinzakka

mohammad200h, 2025-01-23:
Please use  [gmsh_cleaner](https://github.com/mohammad200h/gmsh_cleaner)  to clean gmsh files