google-deepmind/mujoco #1493: mjx.ray with un-normalised vector input gives wrong distances

Andrew-Luo1, 2024-03-10:
Thanks for your work in making this simulator! I think I've found a small bug:

Based on the [description](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#ray-collisions) of mj_ray, it seems like the same distance should result regardless of whether the vector input is normalised. However, I didn't find this to be the case, in the minimal demo below.

```import mujoco
from mujoco import mjx
from functools import partial
import jax 
import jax.numpy as jp

xml_2dof = """
<mujoco model="ball and obstacle">
    <compiler angle="radian"/>
	<option timestep="0.002" iterations="1" ls_iterations="4" solver="Newton" gravity="0 0 -9.81">
		<flag eulerdamp="disable"/>
	</option>
    <worldbody>
        <geom name="obst" size="0.1" type="capsule" fromto="0 2 0 0 2 1" conaffinity="1" contype="0"/>
        <body name="ball" pos="0 0 0.2">
            <geom name="ball_geom" size="0.1" type="sphere" conaffinity="0" contype="1"/>
        </body>
    </worldbody>
    <asset>
        <texture builtin="flat" height="1278" mark="cross" markrgb="1 1 1" name="texgeom" random="0.01" rgb1="0.8 0.6 0.4" rgb2="0.8 0.6 0.4" type="cube" width="127"/>
        markrgb="0.8 0.8 0.8" width="100" height="100"/>
        <material name="geom" texture="texgeom" texuniform="true"/>
    </asset>
</mujoco>
""" 

model = mujoco.MjModel.from_xml_string(xml_2dof)

mx = mjx.put_model(model)
data = mujoco.MjData(model)
mujoco.mj_forward(model, data)
dx = mjx.put_data(model, data)

test_func = jax.jit(partial(mjx.ray, m=mx, d=dx, pnt=jp.zeros(3)))

normalised_vec = jp.array([0, 1, 0])
print("Obstacle distance normalised vec: {}".format(test_func(vec=normalised_vec)[0]))
print("Obstacle distance unnormalised vec: {}".format(test_func(vec=normalised_vec/10)[0]))
```

For the first print, I get 1.9 (correct!); for the second, 19 (off by a factor of 10). 

#### Additional Notes
- mjx.ray also errors out in the above demo if you remove the asset element from the xml; is this expected behaviour?
- The geomgroup argument does not work if you input a jax.numpy.ndarray instead of a tuple of booleans; perhaps this should be supported?
- In case it is of interest, I've used mjx.ray to implement a simple [depth camera](https://github.com/google-deepmind/mujoco/issues/1485); it appears to render a small scene at 3.3e6 fps.

#### Context
OS: Linux
MuJoCo version: 3.1.3
Hardware: CPU

yuvaltassa, 2024-03-10:
Thanks!

While you're at it, could you please check if a similar bug exists with native MuJoCo [ray casting](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#ray-collisions), or if this is an MJX-only bug?

Andrew-Luo1, 2024-03-11:
Glad to help, yes I think I'm finding the same bug with the python binding of mujoco.mj_ray. Using the same model:

```
geomid = np.zeros(1, np.int32)
test_func = partial(mujoco.mj_ray, m=model, d=data, pnt=[0, 0, 0], geomgroup=None, flg_static=1, bodyexclude=-1, geomid=geomid)

normalised_vec = np.array([0, 1, 0])
print("Obstacle distance normalised vec: {}".format(test_func(vec=normalised_vec)))
print("Obstacle distance unnormalised vec: {}".format(test_func(vec=normalised_vec/10)))
```

Again, for the first print, I get 1.9 (correct!); for the second, 19 (off by a factor of 10). 

As a side note, in my opinion, it would be helpful to give defaults for the last four parameters like how mjx.ray does, as all of them are currently required but most users probably simply assume that everything would be hit by the ray.

yuvaltassa, 2024-03-11:
I don't think this is a bug. The [mj_ray docs](https://mujoco.readthedocs.io/en/latest/APIreference/APIfunctions.html#mj-ray)  very clearly define intersection as `(pnt+x*vec, x >= 0)`, so scaling `vec` by the user is legit. Up to you if you want to normalize it. Non-normalized `vec` can be useful if you want to control the units of the reported distance, e.g. to match the units of a hardware sensor. Furthermore, normalization involves a square root, which is not the cheapest operation and we want this code to stay as performant as possible.

I'm going to close this as Works As Intended, but feel free to reply if you disagree.

closed 2024-03-11 by yuvaltassa