google-deepmind/mujoco #1496: SoftBody Dynamics

huanzhizun, 2024-03-11:
Where is the code of SoftBody Dynamics?

quagla, 2024-03-11:
Do you mean the passive forces https://github.com/google-deepmind/mujoco/tree/main/plugin/elasticity ?

huanzhizun, 2024-03-11:
> Do you mean the passive forces https://github.com/google-deepmind/mujoco/tree/main/plugin/elasticity ?

No, it’s the softbody dynamics like it in Bullet: https://github.com/bulletphysics/bullet3/tree/master/src/BulletSoftBody

quagla, 2024-03-11:
Do you mean the length constraints enforcement or the contact resolution?

huanzhizun, 2024-03-13:
> Soft bodies in MuJoCo are treated as bodies connected by constraints. The standard MuJoCo constraint solver is used https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_core_constraint.c#L665 and collisions are solved here https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_collision_driver.c#L1968

Thank you. But which functions are used for that?
 I review the codes and can not find it.

quagla, 2024-03-13:
Can you ask a more precise question? I don't understand what you are looking for.

huanzhizun, 2024-03-13:
> Can you ask a more precise question? I don't understand what you are looking for.

For example, in constraints solve .c file, which functions are used for softbody dynamics. I can not find it so I ask you for help.

quagla, 2024-03-13:
The soft bodies simply have standard mujoco bodies at the vertices, there is no special treatment to describe their dynamics. We add constraints at the edges connecting the vertices and then use the constraint solver in the same way as any other constraint is solved in mujoco. Soft bodies are not a separate concept for us, they are just something that puts together point masses at vertices.

closed 2024-03-14 by quagla