google-deepmind/mujoco #1498: Property as actuator, 'adjustable frictionloss actuator'.

Alesel-9, 2024-03-12:
Hi,

I'm an engineer that are trying to simulate a custom vehicle. In that vehicle i need to adjust the friction loss of a joint as an actuator, since the existning actuaters doesn't acuratly describe the system im working with.
Im wondering if there are such an option or if one can work around it in some way.
Below is a simplified version of the vehicle but i feel it would work as an example.
In it you can see a box (chassie) and four swinging arms attached to it. At the end of the arms are the wheels that are motor driven. I have some temprary actuators in the system already that i want to substitut, they are applied at the site where a want the 'frictionloss actuators'  to be. 

XML
```
<mujoco>
  <compiler autolimits="true" />
  <!-- ASSETS  -->
  <asset>
    <texture name="grid" type="2d" builtin="checker" width="512" height="512" rgb1="1 1 1" rgb2="0.2 0.2 0.2"/>
    <material name="grid" texture="grid" texrepeat="1 1" texuniform="true" reflectance=".2"/>
    <texture type="skybox" builtin="gradient" rgb1="1 1 1" rgb2="0 0 0" width="512" height="512"/>
  </asset>
  <option integrator="implicitfast"/>

  <!-- DEFULTS  -->
  <default>
      <default class="body">
        <geom type="mesh" mesh="body" mass="1"/>
      </default>
      <default class="pendelarm">
        <geom type="mesh" mesh="pendelarm" mass="50"/>
      </default>  
      <default class="hjul">
        <geom type="mesh" mesh="hjul" mass="50"/>
      </default>
      <velocity ctrlrange="0 50"/> 
  </default>
  
  <!-- VISUALS  -->
  <visual>
    <rgba haze="0.15 0.25 0.35 1" contactpoint="0.9 0.6 0.2 1" contactfriction= "0.9 0.8 0.4 1"/>
    <quality shadowsize="4096"/>
    <map stiffness="700" shadowscale="0.5" fogstart="10" fogend="15" zfar="40" haze="0.3"/> 
  </visual>
  
  <!-- WORLD BODIES  -->
  <worldbody>
    <!-- Ground  -->
    <geom type="plane" size="0 0 .01" material="grid" friction="1 1 0.1"/>
    <!-- World Lights  -->
    <light directional="true" diffuse=".4 .4 .4" specular="0.1 0.1 0.1" pos="0 0 5.0" dir="0 0 -1" castshadow="false"/>
    <light directional="true" diffuse=".6 .6 .6" specular="0.2 0.2 0.2" pos="0 0 4" dir="0 0 -1"/>
        
    <!--Car Body -->
    <body pos="0 0 4" euler="0 0 0"  >
      <freejoint/>
      <geom type="box" size="3.5 1. 0.5" rgba=" 0.5 0.3 0 1" />
      <inertial mass="6600" pos="0.3 0 0.8" diaginertia="0.000734064 0.000468438 0.000398719"/>
      <light name="top light" pos="0 0 2" mode="trackcom" diffuse=".4 .4 .4"/>

      <!--Pendel Arm 1-->
      <body name="parm1" pos="-2.8 -1.2 -0.5" euler="0 45 0 ">
          <site name="fls" pos="0.4 0 0"/>
          <joint name="jparm1" type="hinge"  axis="0 1 0" limited="true" range="-45 45"  stiffness="17000"  springref="15" />
          <geom name = "parm1" type="capsule" fromto="0 0 0 0.7 0 0" size="0.07" rgba="0.5 0.2 0 1"/>
          <!--Wheel 1 -->
          <body pos="0.7 -0.1 0" euler="90 0 0"  >
              <joint name="jw1" type="hinge" axis="0 0 1" frictionloss="10" actuatorfrcrange="-0.5 0.5"/>
              <geom name = "w1" type="cylinder"  fromto="0 0 0 0 0 0.25" size="0.35" rgba=".05 .2 .1 1" friction="1 1 0.1"/>
          </body>
      </body>
      
      <!--Pendel Arm 7-->
      <body name="parm7" pos="2. -1.2 -0.5" euler="0 45 0 ">
          <joint name="jparm7" type="hinge"  axis="0 1 0" limited="true" range="-45 45"  stiffness="17000" damping="500" springref="15"/>
          <geom name = "parm7" type="capsule" fromto="0 0 0 0.7 0 0" size="0.07" rgba="0.5 0.2 0 1"/>
          <!--Wheel 7 -->
          <body pos="0.7 -0.1 0" euler="90 0 0"  >
              <joint name="jw7" type="hinge" axis="0 0 1" frictionloss="10"/>
              <geom name = "w7" type="cylinder"  fromto="0 0 0 0 0 0.25" size="0.35" rgba=".05 .2 .1 1" friction="1 1 0.1"/>
          </body>
      </body>
      <!---------------Right side---------------->
      <!--Pendel Arm 8-->
      <body name="parm8" pos="-2.8 1.2 -0.5" euler="0 45 0 ">
          <joint name="jparm8" type="hinge"  axis="0 1 0" limited="true" range="-45 45"  stiffness="17000"  springref="15"/>
          <geom name = "parm8" type="capsule" fromto="0 0 0 0.7 0 0" size="0.07" rgba="0.5 0.2 0 1"/>
          <!--Wheel 8 -->
          <body pos="0.7 0.35 0" euler="90 0 0"  >
              <joint name="jw8" type="hinge" axis="0 0 1" frictionloss="10"/>
              <geom name = "w8" type="cylinder"  fromto="0 0 0 0 0 0.25" size="0.35" rgba=".05 .2 .1 1" friction="1 1 0.1"/>
          </body>
      </body>

      <!--Pendel Arm 14-->
      <body name="parm14" pos="2 1.2 -0.5" euler="0 45 0 ">
          <joint name="jparm14" type="hinge"  axis="0 1 0" limited="true" range="-45 45"  stiffness="17000" damping="500" springref="15" frictionloss="100"/>
          <geom name = "parm14" type="capsule" fromto="0 0 0 0.7 0 0" size="0.07" rgba="0.5 0.2 0 1"/>
          <!--Wheel 14 -->
          <body pos="0.7 0.35 0" euler="90 0 0"  >
              <joint name="jw14" type="hinge" axis="0 0 1" frictionloss="10"/>
              <geom name = "w14" type="cylinder"  fromto="0 0 0 0 0 0.25" size="0.35" rgba=".05 .2 .1 1" friction="1 1 0.1"/>
          </body>
      </body>
    </body>
  </worldbody>
 
  <tendon>
    <fixed name="forward">
      <joint joint="jw1" coef="1"/>
      <joint joint="jw7" coef="1"/>
      <joint joint="jw8" coef="1"/>
      <joint joint="jw14" coef="1"/>
    </fixed>
  </tendon>

  <actuator>
      <damper name="fl" joint="jparm1" kv="1" ctrlrange="0 5000"/>
      <damper name="bl" joint="jparm7" kv="1" ctrlrange="0 5000"/>
      <damper name="fr" joint="jparm8" kv="1" ctrlrange="0 5000"/>
      <damper name="br" joint="jparm14" kv="1" ctrlrange="0 5000"/>
      <velocity name= "move" tendon="forward"/>
  </actuator>

</mujoco>
```

</details>

Here is a screenshot 
[image]

yuvaltassa, 2024-03-12:
Not exactly what you are asking for, but did you try [damper](https://mujoco.readthedocs.io/en/latest/XMLreference.html#actuator-damper) actuators? They do something very similar.

Alesel-9, 2024-03-12:
@yuvaltassa To answer your question of what such an acuater would give me, a regular dampner as implemented in mujoco is calculated by multiplying the velocity of the joint's movement with a scaling constant, as far as i understand the documentation. What i'm looking to implement is to not have it depend on the velocity of the joint, neither its position. I want it to be a constant, but adjustable, resistanse against movement. A constent resistive moment of inertia if you will. 
As for directly adjusting mjModel, I dont quite know what that entails. Can I adjust the model whilst it is runing by changing values in mjModel or am i misunderstanding you? Could you refer to an example or perhaps explain in more detail?

yuvaltassa, 2024-03-12:
> Can I adjust the model whilst it is runing by changing values in mjModel or am i misunderstanding you? 

Of course you can, why not? This is covered in the tutorial.

> Could you refer to an example or perhaps explain in more detail?

You did not say if you are working in Python or C. Assuming Python
```py
model = mj.mjModel( ...
data = mj.mjData(model)
while not_done:
  model.joint(joint_name_or_id).frictionloss = action
  mj.mj_step(model, data)
```

closed 2024-03-12 by yuvaltassa