google-deepmind/mujoco #1500: Feature Request:  faster than realtime simulation in _simulate_

wandida, 2024-03-12:
**Is your feature request related to a problem? Please describe.**
I use the simulate code sample ( https://mujoco.readthedocs.io/en/stable/programming/samples.html#sasimulate ) in Cpp. I did some modifications to support a communication interface with other software. This works very well, as the given code is well structured and commented.

However, I can't do simulations that are faster than realtime. _simulate_ offers the ```realtime``` attribute which can slow down the simulation. E.g. with ```realtime = 0.5``` the simulation is half the speed than realtime but ```realtime = 2``` is not supported.  

**Describe the solution you'd like**
It should work similar to Gazebo ( http://sdformat.org/spec?ver=1.11&elem=physics#physics_real_time_factor ) 
Support of ```realtime``` attribute > 1 in _simulate_. E.g. if ```realtime = 2``` the simulation runs twice as fast as realtime.  
Additionally, if ```realtime = -1``` the simulation should run as fast as possible, limited only by the computational speed.

**Describe alternatives you've considered**
An alternative would be not to  use _simulate_ and build a completely custom simulator with the mujoco Cpp library.

**Additional context**
None

yuvaltassa, 2024-03-12:
We tried faster than realtime simulation, it is not a good interactive UI experience. Human brains don't grok faster than realtime the same way they do slower than realtime. However everything you described is a small and easy change in the simulate codebase, any reason why you can't just modify whatever you need in your fork?

wandida, 2024-03-12:
Hey, thanks for the quick answer :) 

Basically, I tried modifying the ```main.cc``` of _simulate_ (https://github.com/google-deepmind/mujoco/blob/93cbdaee07b7340f6267916ccb412f41ab9efe80/simulate/main.cc#L359) but I didn't get it to work. 

I think allowing this option would benefit the user nevertheless. They will use the UI only for monitoring and not interaction if they use faster than realtime simulation.

yuvaltassa, 2024-03-12:
Just add number greater than one [here](https://github.com/google-deepmind/mujoco/blob/93cbdaee07b7340f6267916ccb412f41ab9efe80/simulate/simulate.h#L215)

closed 2024-03-12 by yuvaltassa

wandida, 2024-03-12:
Thanks, I'll try that right away.
Did you mean greater than 100? The value is in percentage. Most of the numbers are already greater than 1.