google-deepmind/mujoco #1501: Free Joint Euler angle convention for qVel

JeyRunner, 2024-03-12:
Hi,

I'm a student using MuJoCo MJX to learn walking policies with RL.
I want to init the pose of a legged robot (with free joint as root) to a specific pose (with joint positions and velocities).
The format for the free joint in terms of joint velocity is not 100% clear to me.
For a free joint with 7 q values and 6 qVel values:
 - q[0:3] = global xyz position
 - q[3:7] = quaternion rotation
 - qVel[0:3] = global linear velocity
 - qVel[3:6] = angular velocity in the rotated local frame attached to the free joint
 
What is not clear to me is the convention for the angular velocity (qVel[3:6] ), I assume it is expressed as Euler angles obeying the Euler angle convention specified in `compiler/eulerseq` (https://mujoco.readthedocs.io/en/stable/XMLreference.html#compiler-eulerseq) (or does this use some fixed internally defined Euler sequence).

Are these assumptions correct?

yuvaltassa, 2024-03-12:
Velocities are velocities, not angles. Read [this](https://mujoco.readthedocs.io/en/latest/overview.html#floating-objects) (last paragraph).

closed 2024-03-12 by yuvaltassa

JeyRunner, 2024-03-26:
Thanks for the answer, makes sense :)
I mixed up euler angle rates and angular velocities (here the ordering is clear: xyz).