google-deepmind/mujoco #1503: Change MAX_GEOM in _simulate.py

RafaelsNeurons, 2024-03-13:
### Discussed in https://github.com/google-deepmind/mujoco/discussions/1502

<div type='discussions-op-text'>

<sup>Originally posted by **RafaelsNeurons** March 13, 2024</sup>
Hey,
can anyone tell me how I can change MAX_GEOM in _simulate.py ?
I tried to change the value in there bot it does not change and sets itself to 20000 like the original value in this function.

I need to change it, because my simulation crushes because of the Signed Distance Fields and I get a warning to change MAX_GEOM

`    def __init__(self, arg0, arg1, arg2, arg3, arg4, arg5): # real signature unknown; restored from __doc__
        """ __init__(self: mujoco._simulate.Simulate, arg0: object, arg1: object, arg2: object, arg3: object, arg4: bool, arg5: object) -> None """
        pass

    busywait = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ctrl_noise_rate = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ctrl_noise_std = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    dropfilename = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    droploadrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    exitrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    filename = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    load_error = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    measured_slowdown = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    real_time_index = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    refresh_rate = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    run = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    speed_changed = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ui0_enable = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ui1_enable = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    uiloadrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    MAX_GEOM = 20000`</div>

yuvaltassa, 2024-03-13:
What is `_simulate.py`?

RafaelsNeurons, 2024-03-13:
_simulate.py is called by the viewer 

from mujoco import _simulate
_Simulate = _simulate.Simulate

`# imports
import pybind11_builtins as __pybind11_builtins

# functions

def set_glfw_dlhandle(arg0): # real signature unknown; restored from __doc__
    """ set_glfw_dlhandle(arg0: int) -> None """
    pass

# classes

class Mutex(__pybind11_builtins.pybind11_object):
    # no doc
    def __enter__(self): # real signature unknown; restored from __doc__
        """ __enter__(self: mujoco._simulate.Mutex) -> None """
        pass

    def __exit__(self, arg0, arg1, arg2): # real signature unknown; restored from __doc__
        """ __exit__(self: mujoco._simulate.Mutex, arg0: handle, arg1: handle, arg2: handle) -> None """
        pass

    def __init__(self, *args, **kwargs): # real signature unknown
        pass

class Simulate(__pybind11_builtins.pybind11_object):
    # no doc
    def add_to_history(self): # real signature unknown; restored from __doc__
        """ add_to_history(self: mujoco._simulate.Simulate) -> None """
        pass

    def destroy(self): # real signature unknown; restored from __doc__
        """ destroy(self: mujoco._simulate.Simulate) -> None """
        pass

    def exit(self): # real signature unknown; restored from __doc__
        """ exit(self: mujoco._simulate.Simulate) -> None """
        pass

    def load(self, arg0, arg1, arg2): # real signature unknown; restored from __doc__
        """ load(self: mujoco._simulate.Simulate, arg0: object, arg1: object, arg2: str) -> None """
        pass

    def load_message(self, arg0): # real signature unknown; restored from __doc__
        """ load_message(self: mujoco._simulate.Simulate, arg0: str) -> None """
        pass

    def load_message_clear(self): # real signature unknown; restored from __doc__
        """ load_message_clear(self: mujoco._simulate.Simulate) -> None """
        pass

    def lock(self): # real signature unknown; restored from __doc__
        """ lock(self: mujoco._simulate.Simulate) -> mujoco._simulate.Mutex """
        pass

    def render_loop(self): # real signature unknown; restored from __doc__
        """ render_loop(self: mujoco._simulate.Simulate) -> None """
        pass

    def sync(self): # real signature unknown; restored from __doc__
        """ sync(self: mujoco._simulate.Simulate) -> None """
        pass

    def uiloadrequest_decrement(self): # real signature unknown; restored from __doc__
        """ uiloadrequest_decrement(self: mujoco._simulate.Simulate) -> None """
        pass

    def update_hfield(self, arg0): # real signature unknown; restored from __doc__
        """ update_hfield(self: mujoco._simulate.Simulate, arg0: int) -> None """
        pass

    def update_mesh(self, arg0): # real signature unknown; restored from __doc__
        """ update_mesh(self: mujoco._simulate.Simulate, arg0: int) -> None """
        pass

    def update_texture(self, arg0): # real signature unknown; restored from __doc__
        """ update_texture(self: mujoco._simulate.Simulate, arg0: int) -> None """
        pass

    def __init__(self, arg0, arg1, arg2, arg3, arg4, arg5): # real signature unknown; restored from __doc__
        """ __init__(self: mujoco._simulate.Simulate, arg0: object, arg1: object, arg2: object, arg3: object, arg4: bool, arg5: object) -> None """
        pass

    busywait = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ctrl_noise_rate = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ctrl_noise_std = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    dropfilename = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    droploadrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    exitrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    filename = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    load_error = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    measured_slowdown = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    real_time_index = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    refresh_rate = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    run = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    speed_changed = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ui0_enable = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    ui1_enable = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    uiloadrequest = property(lambda self: object(), lambda self, v: None, lambda self: None)  # default

    MAX_GEOM = 1000000

# variables with complex values

__loader__ = None # (!) real value is '<_frozen_importlib_external.ExtensionFileLoader object at 0x00000176B9EFF100>'

__spec__ = None # (!) real value is "ModuleSpec(name='mujoco._simulate', loader=<_frozen_importlib_external.ExtensionFileLoader object at 0x00000176B9EFF100>, origin='C:\\\\Users\\\\z004hpwx\\\\AppData\\\\Local\\\\pypoetry\\\\Cache\\\\virtualenvs\\\\assembly-planning-FCxxoCDl-py3.10\\\\lib\\\\site-packages\\\\mujoco\\\\_simulate.cp310-win_amd64.pyd')"
`

yuvaltassa, 2024-03-13:
Try changing [this line](https://github.com/google-deepmind/mujoco/blob/693e3c66c83c799e326fa465c15b4ed528054cc2/simulate/simulate.h#L49) and rebuilding from source. If that works for you we can increase it for the next release.

RafaelsNeurons, 2024-03-13:
Do you know where I can find this file when I installed mujoco via pip install ?

Thank you very much!

RafaelsNeurons, 2024-03-13:
When I try to execute

MUJOCO_PATH=C:\Users\rp\Documents\mujoco-3.1.3-windows-x86_64\bin MUJOCO_PLUGIN_PATH=C:\Users\rp\Documents\mujoco-3.1.3-windows-x86_64\bin\mujoco_plugin 
pip install mujoco-3.1.4.tar.gz

I get the following error:
RuntimeError: Cannot find MuJoCo library and/or include paths
  [end of output]

**Did I set MUJOCO_PATH and MUJOCO_PLUGIN wrong?**
According to https://mujoco.readthedocs.io/en/stable/python.html  "Build from Source" Chapter it should be right I guess, if I understood point 6 right.

RafaelsNeurons, 2024-03-22:
Unfortunately, changing the value did not help. I have created a bug issue and provided a zip with video, XML and .obj files to reproduce the error. 
Thanks for the answer anyway @yuvaltassa .
 I hope the provided file helps to find the error, maybe until the next release

https://github.com/google-deepmind/mujoco/issues/1539#issue-2202309727

closed 2024-05-05 by yuvaltassa