google-deepmind/mujoco #1511: Provide Official XML Schema for MuJoCo Files

madhephaestus, 2024-03-14:
It would be helpful for bindings developers to have an officially supported set of XML schema files to validated MuJoCo XML files. This repository has the schema through 2.x https://github.com/ronansgd/xml-schema-mjcf

It would be very helpful if these were adopted and kept up to date with any XML changes. 

I am doing my best to add and correct the schema for use in the Java bindings. I will likely maintain a fork, even if it is out of date. 

These schema files could be used by many different language implementations to validate a generated XML file, and used to generate marshaling and unmarshaling code for MuJoCo XML models.

saran-t, 2024-03-18:
Any thoughts on the preferred schema format?

yuvaltassa, 2024-03-18:
duplicate of #6 (that's how long we want to do this! 🙂)

Following up on @saran-t, specifically, there are three obvious candidates, XSD, relaxNG (RNG) and relaxNG-compact (RNC). RNC is by far the most beautiful but also the least commonly supported. AFAICT, both XSD and RNG are fairly widely supported, yet RNG is far more flexible. 

For this to worth our while we would need to replace the struct at the top of [xml_native_reader.cc](https://github.com/google-deepmind/mujoco/blob/main/src/xml/xml_native_reader.cc) as the **source of truth** for the MJCF schema. 

I believe the plan was to 
1. Pick a schema format.
2. Look at available validating XML parsers (e.g. [this one](https://gitlab.gnome.org/GNOME/libxml2/-/wikis/home)), see if any of them is significantly faster/better than tinyXML2. Note that a new, validating XML parser is not a hard requirement. We can just use tinyXML2, since we don't intend to have very much information in the schema file, there are far too many things in MJCF that are not representable by any schema language.
3. Rewrite xml_native_reader to use the schema file rather than the struct.

We got stuck at step 1 🙂 

Closing since #6 is still open, but obviously this is not resolved.

closed 2024-03-18 by yuvaltassa