google-deepmind/mujoco #1513: Sensors in MuJoCo XLA?

eleninisioti, 2024-03-15:
I see in the documentation of MuJoCO XLA (https://mujoco.readthedocs.io/en/stable/mjx.html) that sensors are under development, any chance this feature has been added? If so, are there any examples?

breakds, 2024-03-19:
At this moment, I just implement them with jax functions following the logic implemented in c++: https://github.com/google-deepmind/mujoco/blob/main/src/engine/engine_sensor.c#L525

yuvaltassa, 2024-03-19:
Does anyone want to send us a PR implementing this in the MJX codebase? That would be welcome.

breakds, 2024-03-28:
> Does anyone want to send us a PR implementing this in the MJX codebase? That would be welcome.

Never contributed to MuJoCo before, but I would like to give it a try if no one is currently working on it.

kevinzakka, 2024-08-30:
Sensors will be part of the next MJX release.

closed 2024-08-30 by kevinzakka