google-deepmind/mujoco #1514: Can Mujoco be non-deterministic?

mavex857, 2024-03-15:
Hey, I am using Mujoco in python with gym environments and observed some weird behavior. I am not sure if this is due to Mujoco or the implementation of the environment I am using, so I thought I will just ask here. 
I am observing that when I am in the same state taking the same action, my next state is not identical. I have appended a screenshot to show some concrete numbers. After each ```target```line I reset the environment, my ```action```is always the same so I didn't print it for readability and ```obs``` are the following observations. So from my initial state I take 5 times the same action and observe 5 subsequent state. One can see that the two trajectories are not equal.

I am aware that that could also be due to non-determinism in my environment implementation, but I am not aware of any. So maybe just a quick info if that can stem from Mujoco or if it has to be environment?

[image]

yuvaltassa, 2024-03-15:
https://mujoco.readthedocs.io/en/latest/computation/index.html#reproducibility

closed 2024-03-15 by yuvaltassa

Kallinteris-Andreas, 2024-03-15:
@mavex857 you may want to use `check_mujoco_reset_state` to test your gymnasium environment https://github.com/Farama-Foundation/Gymnasium/blob/main/gymnasium/envs/mujoco/utils.py